RROrg/rr #324: 你好，sa6400机型是不是没法装在nvme的盘上，目前进不去系统

Jie-baobao, 2023-11-04:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  n100
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  23.11.0
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  sa6400
 * version:  7.2

Issue:  ESXi安装进不去系统，能加载引导。提示等待几分钟界面后无论等多久都进不去系统

logs:  

(Plz review the content of #173, #175, #226 first)  
...

kt212878, 2023-11-04:
这里有一篇教程
https://blog.jim.plus/blog/post/jim/synology-installation-with-nvme-disks-only

wjz304, 2023-11-04:
@jim3ma

jim3ma, 2023-11-04:
是想把系统安装到 NVMe，还是引导装反 NVMe 上？

zhaianan, 2023-11-05:
> 是想把群晖系统安装到 NVMe，还是引导装在 NVMe 上？

我将img文件写入nvme盘，X86 P5在启动项不显示这个盘，23.10.5说支持nvme引导，我尝试并不行🥲

wjz304, 2023-11-05:
n100 都去 https://github.com/wjz304/rr/issues/317 吧

closed 2023-11-05 by wjz304