RROrg/rr #332: 板载RTL8111+PCIE RTL8125BG重启后随机调换网口顺序

koihano, 2023-11-05:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  Colorful-BATTLE-AX B360M-E PRO V21
 * CPU:  Intel Core i3-8100
 * NIC:  RTL8111H（Onboard）RTL8125BG（PCI-E）

RR version: (You can find it in the update menu.)  
 * RR:  23.11.0
 * addons:  sortnetif
 * modules:  default
 * lkms:  prod

DSM: 
 * model:  DVA3221
 * version:  7.2.1-69057u1

Issue: 
从DS918+不保留配置重装为DVA3221后，网口默认为8111（NIC1）8125（NIC2）。在不明原因下，重启NAS后会可能导致两张网卡的顺序调换（但设置的静态IP等配置并没有跟随网卡位置的的对调而调换，还是遵循在原来位置上的配置）。尝试使用sortnetif插件将8125设置为NIC1（参数：macs:00e04c680be0），但始终会将8125的位置排在NIC2上，且偶尔会失效造成两张网卡的位置对调。

logs:  
待抓取后补充

wjz304, 2023-11-09:
log 中看到你并未添加 sortnetif插件，或者添加以后并未重新编译使其生效

koihano, 2023-11-11:
> log 中看到你并未添加 sortnetif插件，或者添加以后并未重新编译使其生效

[logs.zip](https://github.com/wjz304/rr/files/13324411/logs.zip)
不好意思，耽搁了几天现在才重新抓logs。目前网卡随机调换问题已确实通过sortnetif插件解决，但填写的参数是【macs:NIC2MAC,NIC1MAC】，即位置是反过来的，在DSM里位置才正常，希望这个bug的发现能给开发者大大提供到一些帮助。

wjz304, 2023-11-11:
把 “macs:” 去掉

koihano, 2023-11-12:
> 把 “macs:” 去掉， 只改 dsm系统，RR下不会改 是已知的，

去掉后正常填写可按顺序识别了，感谢解惑。目前就是需要DSM内固定网卡顺序的功能，RR内的网卡顺序是经过观察应该是固定先识别板载8111后识别8125的，不影响配置使用。只是去掉sortnetif插件后DSM网卡顺序依旧会乱跳。

wjz304, 2023-11-12:
> 只是去掉sortnetif插件后DSM网卡顺序依旧会乱跳。

那肯定的啊，

wjz304, 2023-11-12:
The problem has been resolved,
The issue is closed

closed 2023-11-12 by wjz304