RROrg/rr #339: 23.11.2 и "add wireless"

longtongue91, 2023-11-08:
[image]
Неужели будет возможность подключаться по вай-фай? Я обновился на эту версию, но ничего в меню аддонов не нашёл. Как можно активировать и попробовать эту фичу.
Надеюсь это то, о чём я думаю)

wjz304, 2023-11-08:
Sorry, I wrote the wrong script and the addon was not packaged.
Will be fixed in next version

Yes, you can connect to wifi, but some network cards are supported

wjz304, 2023-11-08:
4.x:
[image]
5.x:
[image]

intel 待测试，

longtongue91, 2023-11-08:
> Yes, you can connect to wifi, but some network cards are supported
Вы мой герой. С нетерпением жду новую версию, что бы попробовать!

closed 2023-11-08 by longtongue91

wjz304, 2023-11-09:
https://github.com/wjz304/rr/releases/tag/23.11.3