RROrg/rr #342: ESXI6.7用arpl-i18n引导安装显示Loader disk not found

RLTyf, 2023-11-10:
### 在#276中看到了相同情况，但是我的是sata硬盘，主板bios硬盘模式是AHCI。
[image]
[image]
虚拟机创建的引导盘1和新建的一个硬盘2都在这个硬盘，引导盘1的控制器用 **SATA控制器0** 显示磁盘类型不受支持无法引导，改为IDE控制器后正常进入引导，但是显示Loader disk not found，把硬盘2控制器改为IDE依旧如此。
[image]
[image]
[image]
[image]

wjz304, 2023-11-10:
不支持 ide， 不受支持，自己研究吧，我不了解 esxi

RLTyf, 2023-11-11:
> 不支持 ide， 不受支持，自己研究吧，我不了解 esxi

好的，谢谢……

closed 2023-11-11 by RLTyf

RLTyf, 2023-11-11:
已解决，原因是把引导转VMDK格式时文件与ESXI不兼容，更新StarWind V2V Converter官网最新版本后就好了