RROrg/rr #346: Virtio相关驱动注入失败？

xrh0905, 2023-11-11:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  QEMU STANDER PC (i440FX + PIIX, 1996)
 * CPU:  AMD Ryzen 9 3900X 12-Core
 * NIC:  vmxnet3
Unraid i440fx-7.1(q35相同故障)

RR version: (You can find it in the update menu.)  
 * RR:  23.11.3
 * addons:  23.11.4
 * modules:  23.11.4
 * lkms:  23.10.0 (prod)

DSM: 
 * model:  DS918+
 * version:  7.2

Issue:  
在Unraid下``和``两个驱动在群晖中都加载失败，`lsmod`看不到，也没有正常生效。
引导时的日志附在下面

logs:  

(Plz review the content of #173, #175, #226 first)  
...
[boot.log.txt](https://github.com/wjz304/rr/files/13325497/boot.log.txt)

wjz304, 2023-11-14:
注入 是指？

wjz304, 2023-11-14:
你放了自己编的？

xrh0905, 2023-11-14:
在模块页面选中 进系统找不到模块 驱动不起来

wjz304, 2023-11-14:
你选了啥，依赖都选了？

xrh0905, 2023-11-14:
全选模块也没用 只选识别到的也不行

xrh0905, 2023-11-18:
```
virtio-pci 0000:00:03.0: virtio_pci: unable to map virtio 56@0 on bar 4
[    5.411295] virtio-pci: probe of 0000:00:03.0 failed with error -22
[    5.426284] vmap allocation for size 70368744185856 failed: use vmalloc=<size> to increase size.
```
不知道是否与这个有关
现在用vmxnet能跑 但是qemu-ga连不上

wjz304, 2023-11-18:
这个问题好像和宿主机的设置有一些关系，
你可以 装原来好使的版本，如果现在也不好使 就是宿主机的问题。

xrh0905, 2023-11-19:
应该是unraid的问题 我在pve上相同配置能正常识别... 真是怪了

wjz304, 2024-03-30:
复现请重新打开

closed 2024-03-30 by wjz304