RROrg/rr #347: BUG

xintonglaile, 2023-11-12:
使用11.4在ESXI8U2安装报错。
[](url)

[image]

xintonglaile, 2023-11-12:
使用23.10.9能正常进入

[image]

xintonglaile, 2023-11-13:
Please fill in the following information.

Install ENV: (You can find it in the boot interface.)

DMI:
CPU:N5105
NIC:
RR version: (You can find it in the update menu.)

RR:23.11.4
addons:
modules:
lkms:
DSM:

model:
version:
Issue:

logs:

xintonglaile, 2023-11-13:
我试了下。11.1和11.2都能正常启动和引导。就是11.3和11.4无法开机也不法编译引导

wjz304, 2023-11-18:
原因不明，尝试换写盘工具吧

xintonglaile, 2023-11-19:
> 原因不明，尝试换写盘工具吧

前几天我也试了虚拟机进PE把rr用写盘工具写入硬盘。开机也是报错。用11.2升级也是报错。就看未来新版本能不能用了。这问题挺奇怪的哦

wjz304, 2024-03-30:
复现请重新打开

closed 2024-03-30 by wjz304