RROrg/rr #348: Exsi 7.0.3使用23.11.4套件会自动停止

koreakid, 2023-11-13:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  N6000
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  23.11.4
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  DSM7.1.1

Issue:  

logs:  

之前使用23.11.1加串口（使用输入文件）就正常，如果换成其他，比如使用物理串行端口，套件内的套件（比如plex）在一段时间后就自动停止了。现在更新到23.11.4，不用加串口也可以使用了。但是出现了套件停止的问题了。

xintonglaile, 2023-11-13:
我直接起不来。开机报错就没然后了

koreakid, 2023-11-13:
> 我直接起不来。开机报错就没然后了

你什么硬件？什么系统？可以加下串口看看

xintonglaile, 2023-11-13:
> > 我直接起不来。开机报错就没然后了> > 我直接起不来。开机报错就没然后了>> 你什么硬件？什么系统？可以加下串口看看>> 你什么硬件？什么系统？可以加下串口看看我的是n5105，esxi8.0u2。等会去试试加个串口看看

xintonglaile, 2023-11-13:
> > 我直接起不来。开机报错就没然后了
> 
> 你什么硬件？什么系统？可以加下串口看看

我试了下。11.1和11.2都能正常启动和引导。就是11.3和11.4无法开机也不法编译引导

koreakid, 2023-11-13:
> > > 我直接起不来。开机报错就没然后了> > 我直接起不来。开机报错就没然后了>> 你什么硬件？什么系统？可以加下串口看看>> 你什么硬件？什么系统？可以加下串口看看我的是n5105，esxi8.0u2。等会去试试加个串口看看

这个要开发回复了。两个差别什么

livebelive, 2023-11-15:
我的套件使用正常啊，我是直接更新到23.11.4的，没有重新下载重新部署，删除了添加的那个串口设备，一切正常，没出现过套件停止的情况，装了几个常用的套件，一个Jellyfin docker nastool 还有相册，都没出现过。

livebelive, 2023-11-15:
我是B760i 12500 SA6400+ 直通i226 4口网卡一张 直通770核显

koreakid, 2023-11-15:
> 我的套件使用正常啊，我是直接更新到23.11.4的，没有重新下载重新部署，删除了添加的那个串口设备，一切正常，没出现过套件停止的情况，装了几个常用的套件，一个Jellyfin docker nastool 还有相册，都没出现过。

也是用exsi吗？我的确实是这样。23.11.4我添加串口还是一样的情况…目前只能降级到23.11.1，加串口，然后我把串口输出文件删了，好像就没了。目前用的时候挺稳定的。暂时就这样吧。

23.11.4能直接驱动12500的核显硬解？不要再加i915的驱动吗？用的是dsm7.1还是7.2

我也有台12500，装pve

livebelive, 2023-11-17:
@koreakid 核显能不能直驱动我不知道，应该是不行的，引导和核显本就是两个概念，核显得找JIM搞得，我本身就是已经搞好了核显的。

livebelive, 2023-11-17:
@koreakid PVE 好像核显可以用吧，不想esxi，没用过PVE 我是DSM7.2

koreakid, 2023-11-17:
> @koreakid PVE 好像核显可以用吧，不想esxi，没用过PVE 我是DSM7.2

@livebelive 稳定吗？我这个不太稳定

livebelive, 2023-11-17:
@koreakid 我找JIM弄个核显驱动，付费搞的，挺稳的，一直在用。