RROrg/rr #349: Exsi使用rr引导DSM7.2，直通HBA卡后，HBA卡中的盘序问题

inbluemoon, 2023-11-13:
Exsi6.5，引导放在虚拟机的SATA控制器 0:0 上面，直通PCIE LSI-2004, 进到DSM中发现HBA中插的两个硬盘序号为 31 32，发现Exsi的SATA控制器有30个Port, 我设置SataPortMap=1, DiskIdxMap=0C， build后硬盘序号为 2 3， 有什么办法让HBA中的硬盘序号从1开始吗

inbluemoon, 2023-11-13:
[image]
[image]

wjz304, 2023-11-13:
U盘引导

inbluemoon, 2023-11-13:
我回去试下Exsi能不能将硬盘指定为USB控制器，如果不行的话，就只能再想想别的办法了

wjz304, 2023-11-13:
sata_remap 也行，但是 不建议，

inbluemoon, 2023-11-13:
sata_remap=0>4:4>0 应该可以解决，能请教一下不建议是因为什么吗

wjz304, 2023-11-13:
> sata_remap=0>4:4>0 应该可以解决，能请教一下不建议是因为什么吗

1. 当你加 硬盘4的时候，你要再该参数吗？
2. 你不觉得这个行为就是指鹿为马吗？

inbluemoon, 2023-11-13:
是这样的，看起来还是需要用USB引导，发生这种情况的原因是因为我用了HBA卡吗，如果直通SATA控制器，就没有这个问题

wjz304, 2023-11-13:
环境比较复杂，所以目前我没有添加相应的设置

inbluemoon, 2023-11-13:
了解了，我回去试一下USB引导

closed 2023-11-13 by inbluemoon