RROrg/rr #351: 升级到23.11.4后，intel X710-T4L网卡丢失。

wu0lei, 2023-11-13:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  Vmware esxi 8.0u1
 * CPU:  E-2146g，
 * NIC:  X710-T4L

RR version: (You can find it in the update menu.)  
 * RR:  23.11.4
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS918+
 * version:  7.2.1 69057

Issue:  升级到23.11.4后，直通的intel X710-T4L网卡丢失，引导内能识别到X710-T4L网卡，但群晖内该网卡未显示。查看模块列表，发现i40e驱动已勾选。
另外：23.11.3版本是正常的。

logs:  

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-11-13:
就是想解下 i40e 报错，取个log看看报啥错

wu0lei, 2023-11-13:
请问大佬 是用哪个编号的教程取日志啊？#173还是#175?

wu0lei, 2023-11-13:
[Uploading serial.log…]()

wu0lei, 2023-11-13:
[serial.log](https://github.com/wjz304/rr/files/13334215/serial.log)

wjz304, 2023-11-13:
有其他网卡能连上？  
```
dmesg | grep i40e
lsmod
ls -l /sys/class/net/*/device/driver 
```

wu0lei, 2023-11-13:
是的 除此以外还有一个vmware的虚拟网卡 VMXNET 3

wjz304, 2023-11-13:
lspci -Qnn

wjz304, 2023-11-13:
dmesg | grep i40e
没有返回任何信息吗？

wjz304, 2023-11-13:
好吧，我知道了，等下个版本。

wu0lei, 2023-11-13:
root@vDSM:~# dmesg | grep i40e
root@vDSM:~#
 确实啥也没返回呢 谢谢

wjz304, 2023-11-13:
因为 2.20.12版本的i40e 一直报错，虽然报错半年了 因为一直没人反馈就没动它，23.11.4 解决一些兼容性问题的时候我就顺手把i40e回退到 1.3.46-k的版本了，刚看了下  1.3.46-k 的i40e 不支持 8086:15ff 的硬件

wjz304, 2023-11-13:
😔，下个版本先回退 2.20.12的i40e吧，报错就先报着吧

wu0lei, 2023-11-13:
好的 后面需要做测试我全程配合 感谢

wu0lei, 2023-11-13:
[image]
另外我说下esxi的i40e版本为：2.5.2.0-1OEM.700.1.0.15843807，另外到intel官网看了下该网卡的linux最新驱动版本是：2.23.17

wjz304, 2023-11-14:
> [image] 另外我说下esxi的i40e版本为：2.5.2.0-1OEM.700.1.0.15843807，另外到intel官网看了下该网卡的linux最新驱动版本是：2.23.17

2.23.17的驱动要生成单独的头文件，这个单独对每个平台单独处理一下子，暂时没时间登闲暇之余吧

wjz304, 2023-11-14:
https://github.com/wjz304/rr-modules/releases/tag/23.11.6

wu0lei, 2023-11-14:
收到 刚到家我更新一下

wu0lei, 2023-11-14:
[image]
更新全部重启后这样了。。。。

wu0lei, 2023-11-14:
喝啥酒了？又换了新版本（23.11.6）好了 X710-T4L网卡能识别了

wjz304, 2023-11-14:
[image]

wjz304, 2023-11-14:
好了，那我 closed了

closed 2023-11-14 by wjz304