RROrg/rr #353: ConnectX-4 Lx网卡无法在918机型中使用

yhysunwei, 2023-11-14:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  Z370 Taichi
 * CPU:  i7 8700T
 * NIC:  Mellanox Technologies MT27710 Family [ConnectX-4 Lx]

RR version: (You can find it in the update menu.)  
 * RR:  v23.11.4
 * addons:  23.11.5
 * modules:  23.11.5
 * lkms:  23.10.0

DSM: 
 * model:  DS918+
 * version:  7.2

wjz304, 2023-11-14:
参考置顶 issue 取 pid vid

yhysunwei, 2023-11-14:
923机型没问题  对比了一下modules   923机型有mlx_compat这个模块
Mellanox Technologies MT27710 Family [ConnectX-4 Lx] [15b3:1015]

ohcocacola, 2023-11-15:
换机型试试，我前段时间918用mcx4121a/sriov不行，换3617能用了！

yhysunwei, 2023-11-16:
> 换支架试用，我前段时间918用mcx4121a/sriov不行，换3617能用了！

主要是要918或者920的硬解

wjz304, 2023-11-16:
https://github.com/wjz304/rr/issues/361

wjz304, 2023-11-16:
dmesg | grep mlx5

yhysunwei, 2023-11-17:
> dmesg | grep mlx5

sunwei@test:/$ dmesg | grep mlx5
[ 12.925245] mlx5_core 0000:03:00.0: enabling device (0000 -> 0002)
[ 12.926115] mlx5_core 0000:03:00.0: firmware version: 14.32.1010
[ 13.661500] mlx5_core 0000:03:00.0: 0000:03:00.0:handle_hca_cap:438:(pid 6016): log_max_qp value in current profile is 18, changing it to HCA capability limit (17)
[ 13.672173] mlx5_core 0000:03:00.0: handle_hca_cap failed
[ 14.059052] mlx5_core 0000:03:00.0: mlx5_load_one failed with error code -22
[ 14.059582] mlx5_core: probe of 0000:03:00.0 failed with error -22

wjz304, 2023-11-18:
#361 跟踪

closed 2023-11-18 by wjz304