RROrg/rr #355: BUG

taohuashui, 2023-11-14:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  b360 m arous pro
 * CPU:  intel i5 8600t
 * NIC:  bcm 57810 \intel i219v\rtl 8852be

RR version: (You can find it in the update menu.)  
 * RR:  11.4
 * addons:  不涉及
 * modules:  不涉及
 * lkms:  不涉及

DSM: 
 * model:  sa6400
 * version:  7.2

Issue:  建议增加8852be无线网卡驱动，这个网卡很便宜，也能用，github有相关资料，可参考https://blog.csdn.net/weixin_47047999/article/details/125127663

logs:  

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-11-14:
rtw89  4.4内核搞不了，最多 在sa6400上搞下

wjz304, 2023-11-14:
dsm 的内核太旧了，新出的东西移植很麻烦

taohuashui, 2023-11-15:
我有918和920的洗白码，没有sa6400洗白码，但是我发现sa6400内核是5.10后我就马上切换到sa6400了，bnx2x_mod
驱动是配合2.5g猫棒使用的，可惜在sa6400上不能正常使用，不知道群晖官方以后发布的型号内核是否会全线升级到5.10，感觉4.4内核寿命不会太长了，不过当前大部分型号还是4.4内核，我没有提交代码的能力，我只能提个建议：是否可以写个判断，让不同驱动匹配对应内核呢？

wjz304, 2023-11-18:
咋判断？

o-i-i-o, 2023-11-18:
我对驱动源代码不懂，不知道为什么在linux上会存在不同内核需要重新编译二进制文件的情况，我以为可以在系统层面加判断脚本啥的，如果不行，那么不通内核的驱动的源代码有何不同？可否把这个判断写到驱动源代码？如果还是不行，可否辛苦帮忙再编译一个sa6400可用的bnx2x_mod驱动。bnx2x为什么又可以在4.4内核和5.10内核都能正常运行呢？

wjz304, 2023-11-19:
报啥错？

o-i-i-o, 2023-11-19:
大师，您掐手算一下吧，DSM的web没有报错，问题出在网卡插上模块以后几秒能识别，过几秒又不能识别，以前arpl10.4没有bnx2x-mod就是这样。ps：我在网卡把1000m速率关闭了，不然是能正常协商1000m的。

closed 2024-03-30 by wjz304