RROrg/rr #356: Не удаётся включить беспроводную сеть

longtongue91, 2023-11-14:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  23.11.4
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS920+
 * version:  DSM 7.2.1-69057 Update 1

Issue:  Не удаётся подключить Wi-Fi. Отсутствует wlan интерфейс. Может я зря думал, что автоматический работает? Нужно ли что-то настроить дополнительно?

logs:  

(Plz review the content of #173, #175, #226 first)  
...

[image]
[image]
[image]

wjz304, 2023-11-14:
There are many RTL models, many with different suffixes may not be supported.
usb network card or pci network card? get the pidvid
```
lsusb
lspci-Qnn
```

longtongue91, 2023-11-14:
> Существует множество моделей RTL, многие из которых с разными суффиксами могут не поддерживаться. Сетевая карта USB или сетевая карта PCI? получить пидвид
> 
> ```
> lsusb
> lspci-Qnn
> ```

[image]
sh-4.4# lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhc i-hcd xHCI Host Controller 0000:00:14.0) hub
  |__1-5         0bda:8179:0000 00  2.00  480MBit/s 500mA 1IF  (Realtek 802.11n  NIC 00E04C0001)
  |__1-10        00da:8510:0100 00  1.10   12MBit/s 50mA 2IFs (Telink Wireless R eceiver TLSR8366)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhc i-hcd xHCI Host Controller 0000:00:14.0) hub
  |__2-7         f400:f400:0100 00  3.20 5000MBit/s 896mA 1IF  ( USB  SanDisk 3. 2Gen1 0501287fe25aaa6069678ba87a080a49ae459132d6b0a75862af223d8ec673b1b48a000000 00000000000000b87357b1ff1d1a1083558107b32cfdb7)
sh-4.4# lspci-Qnn
sh: lspci-Qnn: command not found
sh-4.4#

longtongue91, 2023-11-14:
> lsusb
> lspci-Qnn

Я вставил usb wi-fi, на всякий случай
sh-4.4# |__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhc                                                                                                                                                             i-hcd xHCI Host Controller 0000:00:14.0) hub
sh: syntax error near unexpected token `|'
sh-4.4#   |__1-5         0bda:8179:0000 00  2.00  480MBit/s 500mA 1IF  (Realtek 802.11n                                                                                                                                                              NIC 00E04C0001)
0000:00:1f.0 ISA bridge [0601]: Intel Corporation 200 Sesh: syntax error near unexpected token `|'
sh-4.4#   |__1-6         148f:7601:0000 00  2.01  480MBit/s 160mA 1IF  (MediaTek 802.11                                                                                                                                                              n WLAN 1.0)
sh: syntax error near unexpected token `|'
sh-4.4#   |__1-10        00da:8510:0100 00  1.10   12MBit/s 50mA 2IFs (Telink Wireless R                                                                                                                                                             eceiver TLSR8366)
sh: syntax error near unexpected token `|'
sh-4.4# |__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhc                                                                                                                                                             i-hcd xHCI Host Controller 0000:00:14.0) hub
sh: syntax error near unexpected token `|'
sh-4.4#   |__2-7         f400:f400:0100 00  3.20 5000MBit/s 896mA 1IF  ( USB  SanDisk 3.                                                                                                                                                             2Gen1 0501287fe25aaa6069678ba87a080a49ae459132d6b0a75862af223d8ec673b1b48a000000                                                                                                                                                             00000000000000b87357b1ff1d1a1083558107b32cfdb7)
sh: syntax error near unexpected token `|'
sh-4.4# sh-4.4# lspci -Qnn
sh: sh-4.4#: command not found
sh-4.4# 0000:00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6t                                                                                                                                                             h Gen Core Processor Host Bridge/DRAM Registers [8086:191f] (rev 07)
sh: syntax error near unexpected token `('
sh-4.4# 0000:00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 530                                                                                                                                                              [8086:1912] (rev 06)
sh: syntax error near unexpected token `('
sh-4.4# 0000:00:14.0 USB controller [0c03]: Intel Corporation 200 Series/Z370 Chipset Fa                                                                                                                                                             mily USB 3.0 xHCI Controller [8086:a2af]
sh: 0000:00:14.0: command not found
sh-4.4# 0000:00:16.0 Communication controller [0780]: Intel Corporation 200 Series PCH C                                                                                                                                                             SME HECI #1 [8086:a2ba]
sh: 0000:00:16.0: command not found
sh-4.4# 0000:00:17.0 SATA controller [0106]: Intel Corporation 200 Series PCH SATA contr                                                                                                                                                             oller [AHCI mode] [8086:a282]
sh: 0000:00:17.0: command not found
sh-4.4# 0000:00:1b.0 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Roo                                                                                                                                                             t Port #17 [8086:a2e7] (rev f0)
sh: 0000:00:1b.0: command not found
sh-4.4# 0000:00:1c.0 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Roo                                                                                                                                                             t Port #1 [8086:a290] (rev f0)
sh: 0000:00:1c.0: command not found
sh-4.4# 0000:00:1c.3 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Roo                                                                                                                                                             t Port #4 [8086:a293] (rev f0)
sh: 0000:00:1c.3: command not found
sh-4.4# 0000:00:1c.7 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Roo                                                                                                                                                             t Port #8 [8086:a297] (rev f0)
sh: 0000:00:1c.7: command not found
sh-4.4# 0000:00:1d.0 PCI bridge [0604]: Intel Corporation 200 Series PCH PCI Express Roo                                                                                                                                                             t Port #9 [8086:a298] (rev f0)
sh: 0000:00:1d.0: command not found
[... cut ...] xpress Gigabit Ethernet Controller [10ec:8168] (rev 15)
sh: syntax error near unexpected token `('
sh-4.4# 0001:01:00.0 SATA controller [0106]: Marvell Technology Group Ltd. 88SE9235 PCIe                                                                                                                                                              2.0 x2 4-port SATA 6 Gb/s Controller [1b4b:9235] (rev 11)
sh: syntax error near unexpected token `('
sh-4.4# sh-4.4#
sh: sh-4.4#: command not found
sh-4.4#

wjz304, 2023-11-14:
0bda:8179   Realtek 802.11n NIC 00E04C0001
148f:7601    MediaTek 802.11n WLAN 1.0
00da:8510   Telink Wireless Receiver TLSR8366

wjz304, 2023-11-14:
It seems that these two models are not supported at present. I will check the source code later and add, and @ you

wjz304, 2023-11-14:
当前的 rtl 驱动
[image]

longtongue91, 2023-11-15:
Загрузился со вставленными usb Wi-Fi. Ничего не изменилось. Интерфейс не появился. И вывод "lspci -Qnn" устройства WiFi не отображает. Я же правильно понял, что эта команда показывает устройства с установленными драйверами?
[image]
@wjz304

wjz304, 2023-11-15:
https://github.com/wjz304/rr-modules/actions/runs/6874721412
[image]
@longtongue91

longtongue91, 2023-11-15:
> https://github.com/wjz304/rr-modules/actions/runs/6874721412

Я не понимаю что я должен сделать по этой ссылке?

wjz304, 2023-11-16:
Download the attachment and connect to RR through SSH

Then upload it in the update menu ->local update.

wjz304, 2023-11-16:
Okay, let me send a version

wjz304, 2023-11-16:
https://github.com/wjz304/rr-modules/releases/tag/23.11.7
update modules

longtongue91, 2023-11-17:
Я обновил модули, но ничего не изменилось.
[image]

longtongue91, 2023-11-17:
Решил преустановить на всякий случай. Теперь при установке `0bda:8179 Realtek 802.11n NIC 00E04C0001 rtl8188eu` dsm зависает. Я нашёл логи, но не могу разобрать куда смотреть. Как я уже говорил, я не очень умный.
[logs.tar.gz](https://github.com/wjz304/rr/files/13392380/logs.tar.gz)
[image]

wjz304, 2023-11-18:
Hey, wifi is so troublesome. There are too many dependencies and it is very troublesome to solve.
I'll study it for a few days first

wjz304, 2023-11-18:
https://github.com/wjz304/rr/issues/362

wjz304, 2025-02-17:
v25.2.1 test and report

closed 2025-02-17 by wjz304