RROrg/rr #357: LSI SAS 9211-8i (HBA) not detected

Ostap6, 2023-11-15:
RR version:   
 * RR:  23.11.6
 * addons:  all
 * modules:  all
 * lkms:  prod

DSM: 
 * model:  RS3621xs+
 * version:  7.2
 
Good day!
How to install driver LSI SAS 9211-8i

wjz304, 2023-11-15:
it model ?

wjz304, 2023-11-16:
Support "it mode" based on feedback from others.

closed 2023-11-16 by wjz304

Ostap6, 2023-11-17:
Found SCSI/HBA "04:00.0 LSI Logic / Symbios Logic MegaRAID SAS 2008 [Falcon] (rev 03)"

wjz304, 2023-11-18:
https://www.kente.tech/2020/12/09/flashing-lsi-2008-hba-to-it-mode/

Ostap6, 2023-11-18:
Thank you! I'll try and report back!

wjz304, 2023-11-18:
Pay attention to the name inside. This URL may be 2308. You'd better look at other instructions.
Because I have difficulty reading English, I didn’t find much relevant information.