RROrg/rr #358: 关于静态IP选项的建议

Shuazijun, 2023-11-15:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  ？？??
 * CPU:  7735HS
 * NIC:  virtio

RR version: (You can find it in the update menu.)  
 * RR:  23.11.6
 * addons:  默认
 * modules:  默认
 * lkms:  默认

DSM: 
 * model:  DS918+
 * version:  7.2

Issue:  关于静态IP设置选项的建议

logs:  None

...

如题，关于高级设置中静态IP设置的问题  
请问能否添加网关设置、DNS设置的选项?  
针对部分无DHCP的网络，且子网掩码不是/24的公网IP，需要手动设置网关及DNS，目前能在菜单上设置的内容不够完整，需要手改系统配置文件来获取网络。

wjz304, 2023-11-16:
UI不好设计，我研究研究吧，

cnbb002, 2023-11-16:
修改哪的系统配置文件来获取网络呢？正好在做测试，目前的网络环境是纯手动配置，自动获取是一个无法联网的段，求指导

wjz304, 2023-11-16:
用ip命令

Shuazijun, 2023-11-16:
IP地址可以根据ip addr指令获取网卡名称  
在通过 nano /etc/network/interfaces 编辑网卡信息，编辑完后重启网络服务  
还有DNS通过  nano /etc/resolv.conf  可以编辑，也需要重启网络服务

wjz304, 2023-11-16:
没那么简单，还要保存

cnbb002, 2023-11-16:
> IP地址可以根据ip addr指令获取网卡名称 在通过 nano /etc/network/interfaces 编辑网卡信息，编辑完后重启网络服务 还有DNS通过 nano /etc/resolv.conf 可以编辑，也需要重启网络服务

可否贴个完整的配置？网卡名称eth0

Shuazijun, 2023-11-16:
参考链接：[CSDN](https://blog.csdn.net/weixin_38090079/article/details/131676737)  
大同小异，自己试下就好了

cnbb002, 2023-11-16:
重启网络的命令是/etc/init.d/network restart么？

Shuazijun, 2023-11-16:
问他 @wjz304 arpl引导编译的环境是哪个发行版，你按发行版去搜就行了。

cnbb002, 2023-11-16:
> 没那么简单，还要保存

23.11.6这个版本是哪个发行环境呢？@wjz304

cnbb002, 2023-11-16:
root@rr:/# cat /etc/network/interfaces
# interface file auto-generated by buildroot

auto lo
iface lo inet loopback

auto eth0
iface eth0 inet static
address 128.5.254.4
netmask 255.255.255.0
gateway 128.5.254.254
root@rr:/# cat /etc/resolv.conf
# Generated by dhcpcd from eth0.dhcp
# /etc/resolv.conf.head can replace this line
nameserver 114.114.114.114
# /etc/resolv.conf.tail can replace this line
root@rr:/# systemctl restart networking
-bash: systemctl: command not found
root@rr:/# /etc/init.d/resolvconf restart
-bash: /etc/init.d/resolvconf: No such file or directory
尝试修改网络配置，但没有合适的网络重启命令，感觉没有生效目前？

wjz304, 2023-11-16:
buildroot  非发行版linux

cnbb002, 2023-11-17:
> buildroot 非发行版linux

那我该用哪个命令对修改的网络进行重启服务呢？如果可以，请指导一下。谢谢！

wjz304, 2023-11-18:
364 跟踪

closed 2023-11-18 by wjz304