RROrg/rr #360: 请问最新的23.11.6版本 SA6400支持12代CPU（i5-12400）的核显硬解吗

wulinhui80, 2023-11-15:
请问最新的23.11.6版本 SA6400支持12代CPU（i5-12400）的核显硬解吗?我看到说支持12代CPU，安装后还是不能调用核显硬解，是不是安装时要加载什么插件吗？

FelixLAI85, 2023-11-16:
> 请问最新的23.11.6版本 SA6400支持12代CPU（i5-12400）的核显硬解吗?我看到说支持12代CPU，安装后还是不能调用核显硬解，是不是安装时要加载什么插件吗？

我的是i312100，确认支持，我把所有模块都选上了，没有安装特定的插件。用的nyanmisaka最新的jellyfin docker，可以硬解。

jim3ma, 2023-11-16:
新驱动，可能兼容性不好，建议上传内核日志分析

Danborad, 2023-11-16:
畅网第四版5105，物理机安装7.2核显显示没问题，但是nyanmisaka最新的jellyfin docker和emby开心版都不能核显解码

wulinhui80, 2023-11-16:
N5105 不支持吧

wulinhui80, 2023-11-16:
> 我是i312100，确认支持，我把所有模块都选上了，没有安装特定的插件。用nyanmisaka最新的jellyfin docker，可以硬解。

是全部的模块都选上？我看了有好多是可以自已选择的。但里面不知道那个是有关于核显的。

FelixLAI85, 2023-11-17:
> > 我是i312100，确认支持，我把所有模块都选上了，没有安装特定的插件。用nyanmisaka最新的jellyfin docker，可以硬解。
> 
> 是全部的模块都选上？我看了有好多是可以自已选择的。但里面不知道那个是关于核显的。
我就是不知道怎么选，所以全选了。。。，然后进系统之后，用ls /dev/dri 命令显示：by-path  card0  renderD128，在 jellyfin里播放4k高清视频的时候，选择1080p可以顺利播放，播放信息里可以看到转码信息，同时cpu占用率很低，之前没有硬解的时候，CPU使用率在转码的时候会飙到90%以上，所以我认为应该是成功开启硬解了。不过目前使用的jellyfin版本使用硬解之后，有些视频播放会有问题，换成emby 4.8.0.58测试版会好一些。

darkgetter, 2023-11-17:
目前安裝使用的是ds920+，請問直接安裝sa6400會否保留到數據？
will the data be kept if i directly install SA6400 over DS920+?

BenBenPeng, 2023-11-18:
零刻 EQ12 N100 双网卡小主机  物理机安装 rr11.6  

root@Synology:~# ls /dev/dri
ls: cannot access '/dev/dri': No such file or directory

emby 也开启不了硬解码

wulinhui80, 2023-11-18:
> > 请问最新的23.11.6版本 SA6400支持12代CPU（i5-12400）的核显硬解吗?我看到说支持12代CPU，安装后还是不能调用核显硬解，是不是安装时要加载什么插件吗？
> 
> 我的是i312100，确认支持，我把所有模块都选上了，没有安装特定的插件。用的nyanmisaka最新的jellyfin docker，可以硬解。

我专门买了个12100来安装，但是还是不能硬解，用ls /dev/dri 命令，没有显示驱动了核显

BenBenPeng, 2023-11-19:
root@Synology:~# ls /dev/dri
by-path  card0  renderD128

root@Synology:~# cat /sys/kernel/debug/dri/0/i915_frequency_info
PM MASK=0x80000000
pm_intrmsk_mbz: 0x00000000
RPSTAT1: 0x0000605d
RPNSWREQ: 200MHz
Lowest (RPN) frequency: 200MHz
Nominal (RP1) frequency: 400MHz
Max non-overclocked (RP0) frequency: 750MHz
Current freq: 200 MHz
Actual freq: 200 MHz
Min freq: 200 MHz
Boost freq: 750 MHz
Max freq: 750 MHz
efficient (RPe) frequency: 400 MHz
Current CD clock frequency: 179200 kHz
Max CD clock frequency: 652800 kHz
Max pixel clock frequency: 1305600 kHz

我这个是显示驱动了核显了吧  但是 emby 用网页播放 依然是 cpu 占用 90%  并且在 emby 里高级里不显示解码器

Promix953, 2023-11-19:
PVE8+N5105
23.11.6,SA6400,DSM7.2
jellyfin内可以硬解，但是会每3秒"GPU HANG"错误导致ffmpeg无法持续运行
我已经给上游i915_backports提交了PR修复 https://github.com/MoetaYuko/intel-gpu-i915-backports/pull/1 
[image]
[image]
可以稳定硬解

FelixLAI85, 2023-11-19:
> > > 请问最新的23.11.6版本 SA6400支持12代CPU（i5-12400）的核显硬解吗?我看到说支持12代CPU，安装后还是不能调用核显硬解，是不是安装时要加载什么插件吗？
> > 
> > 
> > 我的是i312100，确认支持，我把所有模块都选上了，没有安装特定的插件。用的nyanmisaka最新的jellyfin docker，可以硬解。
> 
> 我专门买了个12100来安装，但是还是不能硬解，用ls /dev/dri 命令，没有显示驱动了核显

不知道是不是驱动编译结构有什么问题，我必须把所有模块全选上才可以正确驱动核显，如果我去掉勾选某个驱动，比如网卡驱动，那我的核显也会驱动不了。可能目前的驱动还不是很完善，我建议你把所以驱动模块都选上，然后编译，再启动试试。我目前用jellyfin和emby都可以硬解。我是ESXi8U2, SA6400, 7.2.1 update2。

vikii0210, 2023-11-20:
> 目前安裝使用的是ds920+，請問直接安裝sa6400會否保留到數據？ will the data be kept if i directly install SA6400 over DS920+?

应该是可以的，我之前也是用的ds920+，23.9.0 然后是用的23.10.4升级到sa6400，然后现在又升级到23.11.6 也是sa6400都没有问题

wulinhui80, 2023-11-20:
> root@Synology:~# ls /dev/dri by-path card0 renderD128
> 
> root@Synology:~# cat /sys/kernel/debug/dri/0/i915_frequency_info PM MASK=0x80000000 pm_intrmsk_mbz: 0x00000000 RPSTAT1: 0x0000605d RPNSWREQ: 200MHz Lowest (RPN) frequency: 200MHz Nominal (RP1) frequency: 400MHz Max non-overclocked (RP0) frequency: 750MHz Current freq: 200 MHz Actual freq: 200 MHz Min freq: 200 MHz Boost freq: 750 MHz Max freq: 750 MHz efficient (RPe) frequency: 400 MHz Current CD clock frequency: 179200 kHz Max CD clock frequency: 652800 kHz Max pixel clock frequency: 1305600 kHz
> 
> 我这个是N100 显示驱动了核显了吧 但是 emby 用网页播放 依然是 cpu 占用 90% 并且在 emby 里高级里不显示解码器
> 
> 我一开始是升级 11.6 显示不出来装上了驱动 然后我重新下载写盘 11.6 就可以了 奇怪的是无线网卡intel AX201还是不行。
> 
> 总是显示没有 wlan0 这个接口
是需要重新写盘然后安装的。升级到11.6 就是不能识别核显。

wjz304, 2023-11-20:
wifi 现在有问题，先不要试了

baogenwang, 2023-11-22:
> > > > 请问最新的23.11.6版本SA6400支持12代CPU（i5-12400）的核显硬解吗？我看到说支持12代CPU，安装后还是不能调用核显硬解，是不是安装时要加载什么插件吗？
> > > 
> > > 
> > > 我是i312100，确认支持，我把所有模块都选上了，没有安装特定的插件。用nyanmisaka最新的jellyfin docker，可以硬解。
> > 
> > 
> > 我专门买了个12100来安装，但是还是不能硬解，用ls /dev/dri命令，没有显示驱动了核显
> 
> 不知道是不是驱动编译结构有什么问题，我必须把所有模块全选上才可以正确驱动核显，如果我取消选中某些驱动，比如终止驱动，那我的核显也驱动不了。目前的驱动还不是很完善，我建议你把所以驱动模块都选上，然后编译，再启动尝试。我目前用jellyfin和emby都可以硬解。我是ESXi8U2, SA6400, 7.2.1 update2。

我也驱动不了。我是在UNRAID下面。这个驱动有问题。

baogenwang, 2023-11-22:
> > > > > 请问最新的23.11.6版本SA6400支持12代CPU（i5-12400）的核显硬解吗？我看到说支持12代CPU，安装后还是不能调用核显硬解，是不是安装时要加载什么插件吗？
> > > > 
> > > > 
> > > > 我是i312100，确认支持，我把所有模块都选上了，没有安装特定的插件。用nyanmisaka最新的jellyfin docker，可以硬解。
> > > 
> > > 
> > > 我专门买了个12100来安装，但是还是不能硬解，用ls /dev/dri命令，没有显示驱动了核显
> > 
> > 
> > 不知道是不是驱动编译结构有什么问题，我必须把所有全选上才可以正确驱动核显，如果我取消选中某些驱动，比如终止驱动，那我的核显也驱动不了。还不是很完善，我建议你把所以驱动模块都选上，然后编译，启动再尝试。我目前用jellyfin和emby都可以硬解。我是ESXi8U2, SA6400, 7.2.1 update2。
> 
> 我也驱动不了。我在UNRAID下面SA6400。这个驱动有问题。

wjz304, 2023-11-22:
不兼容 unraid vgpu

lbcz-tian, 2023-11-28:
unraid sr-iov虚拟的核显能用吗🤔

closed 2024-03-30 by wjz304

wjz304, 2024-11-04:
https://github.com/RROrg/rr/issues/3726