RROrg/rr #361: ConnectX-4 Lx网卡安装sa6400无法获取ip

gasment, 2023-11-16:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  （qemu）
 * CPU:  13-8100T
 * NIC:  直通Mellanox ConnectX-4 Lx (MCX4121A-ACAT)

RR version: (You can find it in the update menu.)  
 * RR:  23.11.6
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  sa6400
 * version:  7.2.1-69057 Update 2

Issue:  
在进入DSM引导后的串口控制台中可以查询到驱动已被加载，但是没有ifconfig中没有对应的eth；
使用VirtIO虚拟网卡可正常获取ip并引导进系统；
尝试过918+/920+，问题相同
[image]
logs:  
网卡pid&vid：Mellanox Technologies MT27710 Family [ConnectX-4 Lx] [15b3:1015]

gasment, 2023-11-16:
目前编译机型1823XS+，正常使用该网卡

wjz304, 2023-11-16:
dmesg | grep mlx5

gasment, 2023-11-16:
#rr内：
root@rr:/opt/rr# dmesg | grep mlx5
[    9.550101] mlx5_core 0000:06:10.0: firmware version: 14.21.2030
[    9.550111] mlx5_core 0000:06:10.0: 0.000 Gb/s available PCIe bandwidth (Unknown x255 link)
[   10.107668] mlx5_core 0000:06:10.0: Port module event: module 0, Cable plugged
[   10.114711] mlx5_core 0000:06:10.0: MLX5E: StrdRq(0) RqSz(1024) StrdSz(256) RxCqeCmprss(0)
[   10.400529] mlx5_core 0000:06:10.0 eth0: Link up
root@rr:/opt/rr# 

#sa6400引导内：
SynologyNAS> dmesg | grep mlx5
[    5.441826] mlx5_core 0000:06:10.0: firmware version: 14.21.2030
[    5.445150] mlx5_core 0000:06:10.0: 0.000 Gb/s available PCIe bandwidth (Unknown x255 link)
[    6.038409] mlx5_core 0000:06:10.0: Port module event: module 0, Cable plugged
SynologyNAS> 

#1823xs+正常进入系统：
root@DSM:~# dmesg | grep mlx5
[    7.293171] mlx5_core 0000:06:1c.0: firmware version: 14.21.2030
[    8.309989] mlx5_core 0000:06:1c.0: Port module event: module 0, Cable plugged
[    8.324676] mlx5_core 0000:06:1c.0: MLX5E: StrdRq(1) RqSz(8) StrdSz(64) RxCqeCmprss(0)
[   19.845286] mlx5_core 0000:06:1c.0 eth0: mlx5e_update_carrier:142: Link up

wjz304, 2023-11-16:
q 我 todesk

wjz304, 2023-11-16:
[ 5.445150] mlx5_core 0000:06:10.0: 0.000 Gb/s available PCIe bandwidth (Unknown x255 link)
看这句 后几行

wjz304, 2023-11-16:
有没有error 啥的

wjz304, 2023-11-16:
dmesg | grep -A 20 'mlx5_core 0000:06:10.0: 0.000 Gb/s'

gasment, 2023-11-16:
SynologyNAS> dmesg | grep -A 20 'mlx5_core 0000:06:10.0: 0.000 Gb/s'
[    5.531079] mlx5_core 0000:06:10.0: 0.000 Gb/s available PCIe bandwidth (Unknown x255 link)
[    5.604850] usbcore: registered new interface driver usbhid
[    5.608978] usbhid: USB HID core driver
[    6.098050] mlx5_core 0000:06:10.0: Port module event: module 0, Cable plugged
[   16.117524] printk: udevd: 13 output lines suppressed due to ratelimiting
[   16.162735] synobios open /dev/ttyS1 success
[   16.163701] Fail to get disk 1 led type
[   16.164548] Fail to get disk 2 led type
[   16.165291] Fail to get disk 3 led type
[   16.166041] Fail to get disk 4 led type
[   16.166804] Fail to get disk 5 led type
[   16.167556] Fail to get disk 6 led type
[   16.168266] Fail to get disk 7 led type
[   16.169048] Fail to get disk 8 led type
[   16.169828] Fail to get disk 9 led type
[   16.170631] Fail to get disk 10 led type
[   16.171354] Fail to get disk 11 led type
[   16.172105] Fail to get disk 12 led type
[   16.172978] 2023-11-16 16:45:21 UTC
[   16.173722] synobios: load, major number 201
[   16.174506] Brand: Synology
SynologyNAS>

完整log:
[dmesg.txt](https://github.com/wjz304/rr/files/13380396/dmesg.txt)

wjz304, 2023-11-16:
没有报错信息， 不好搞啊，明天看看有没有其他版本吧

gasment, 2023-11-17:
对比正常引导的日志，缺少了mlx_compat模块的信息，sa6400的模块包里没有mlx_compat，或许是缺少这个模块导致无法link up?
####dmesg | grep ml
**[    6.780126] Compat-mlnx-ofed backport release: c22af88
[    6.781296] Backport based on mlnx_ofed/mlnx-ofa_kernel-4.0.git c22af88
[    6.782549] compat.git: mlnx_ofed/mlnx-ofa_kernel-4.0.git**
[    6.787992] mlx5_core 0000:06:1c.0: firmware version: 14.21.2030
[    7.800778] mlx5_core 0000:06:1c.0: Port module event: module 0, Cable plugged
[    7.815708] mlx5_core 0000:06:1c.0: MLX5E: StrdRq(1) RqSz(8) StrdSz(64) RxCqeCmprss(0)
[   19.590166] mlx5_core 0000:06:1c.0 eth0: mlx5e_update_carrier:142: Link up
[   35.615977] [16400]: ocs_fc_lio: ocs_driver_print_module_params:1233::  ramlog_size = 1048576

yhysunwei, 2023-11-17:
> dmesg | grep -A 20 'mlx5_core 0000:06:10.0: 0.000 Gb/s'

我之前也反馈过对比  918+和923，923比918多了mlx_compat就正常

wjz304, 2023-11-18:
https://github.com/wjz304/rr/issues/353

wjz304, 2023-11-18:
https://github.com/wjz304/rr-modules/actions/runs/6914618053
有时间试一下

gasment, 2023-11-18:
已在rr引导更新选项上传该modules.zip并成功解压，重新选择型号编译后启动，串口相关日志与之前没有变化，问题依旧

wjz304, 2023-11-19:
不好使，就排着吧

Lstions, 2024-04-03:
好像是因为少了mlx_compat内核模块
https://enterprise-support.nvidia.com/s/article/mellanox-linux-driver-modules-relationship--mlnx-ofed-x
[image]

Lstions, 2024-04-03:
@wjz304

Lstions, 2024-04-03:
> 好像是因为少了mlx_compat内核模块 https://enterprise-support.nvidia.com/s/article/mellanox-linux-driver-modules-relationship--mlnx-ofed-x ![rtaImage](https://private-user-images.githubusercontent.com/37234704/319147731-e2775583-cd8d-4284-9218-cbaf2b75a994.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIxNDA5MjIsIm5iZiI6MTcxMjE0MDYyMiwicGF0aCI6Ii8zNzIzNDcwNC8zMTkxNDc3MzEtZTI3NzU1ODMtY2Q4ZC00Mjg0LTkyMTgtY2JhZjJiNzVhOTk0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAzVDEwMzcwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM3ZmE3N2ZlZTMxYzE5ZmJkYzE5NWQxZTMwNDkwZmQ0ZmJhYTJjMWM3OTdlMjA2NmIwYjcyNWJkYmRiYjg3YTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XW4iJxaKUb2RyTL-I7oKEi2dGd24S8nxwi62z8-xjfY)

@wjz304

Lstions, 2024-04-05:
> 需要移植，wjz304/mlnx-ofed-kernel (github.com)  Ing ***@***.***  
> […](#)

已经验证过正常工作了，是要依赖mlxfw内核模块。我用sa6400的toolchain编译了mlxfw、mlx5_core、xc_dec模块，insmod上去就好了。
[image]

geli-ans, 2024-04-06:
此问题困扰很多天 ，查询资料发现白群 25Gb 卡不支持SA6400 , dmesg mlx5 全部加载 ，查询官网页面，发现 SA3600 同时支持 25Gb CX4 + SAS RAID 3108。替换型号后全部可用。但12代CPU的集显又不行。 

2024-4-6
今日insmod mlxfw、mlx5_core、xc_dec 后，SA6400全部可用（SAS RAID3108、25Gb CX4、12代集显）

wjz304, 2024-04-06:
https://ttttt.link/f/6611625b83e45
编了一个，测试下吧，最好全架构的型号都试一下，
RR shell 下 
export DEBUG=1
然后即可在更新里更新 modules

wjz304, 2024-04-06:
@geli-ans @Lstions @gasment @yhysunwei

gasment, 2024-04-06:
> https://ttttt.link/f/6611625b83e45 编了一个，测试下吧，最好全架构的型号都试一下， RR shell 下 export DEBUG=1 然后即可在更新里更新 modules

测试之前出现问题的SA6400/918+/920+，现已正常使用

wjz304, 2024-04-07:
https://github.com/RROrg/rr/releases/tag/24.4.1

closed 2024-04-07 by wjz304

musk-1024, 2024-04-14:
更新rr 24.4.1前，3622xs+原本可以正常使用mlx5，更新后mod中找不到与mlx5相关项，dmesg| grep mlx5也没有任何输出
下图为更新前所包含的mod
[image]

wjz304, 2024-04-14:
vid pid

> 更新rr 24.4.1前，3622xs+原本可以正常使用mlx5，更新后mod中找不到与mlx5相关项，dmesg| grep mlx5也没有任何输出 下图为更新前所包含的mod [...]

musk-1024, 2024-04-15:
vid : 0x0001
pid : 0x46f4
rr : 24.4.1-24.4.3
DSM version : 7.2(69057)
NIC : ConnectX-5

musk-1024, 2024-04-15:
vid : 15b3:1018
pid : Mellanox Technologies MT27800 Family [ConnectX-5 Virtual Function]
rr : 24.4.1-24.4.3
DSM version : 7.2(69057)
NIC : ConnectX-5

geli-ans, 2024-04-16:
> vid : 15b3:1018 pid : Mellanox Technologies MT27800 Family [ConnectX-5 Virtual Function] rr : 24.4.1-24.4.3 DSM version : 7.2(69057) NIC : ConnectX-5

建议换个机型重新编译，或者回退引导。 机型的话 SA6400 各方面都很优秀了，阵列卡、网卡、NVMe 、 盘位盘序等都好得很，内核还新。