RROrg/rr #362: 无线网卡mt7601un无法驱动

o5rhaxzc, 2023-11-17:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  23.11.6
 * addons:  23.11.5
 * modules:  
 * lkms:  23.10.0

DSM: 
 * model:  DS920+
 * version:  7.2(69057)

Issue:  无线网卡mt7601un无法驱动，dmesg也有一条错误

logs:  

(Plz review the content of #173, #175, #226 first)  
...

[image]

wjz304, 2023-11-17:
DSM shell root
```
curl -kLO https://raw.githubusercontent.com/wjz304/rr-modules/main/thirdparty/firmware/mt7601u.bin
mv -f mt7601u.bin /lib/firmware/mt7601u.bin
```
重启

o5rhaxzc, 2023-11-17:
dsm shell看dmesg还是错误的
[image]

wjz304, 2023-11-17:
放进去重启了？

wjz304, 2023-11-17:
等我直接更新的吧

o5rhaxzc, 2023-11-17:
替换了 我怕文件下载出错还校验过md5sum 还是驱动不了 Interface wlan0 not found

wjz304, 2023-11-18:
#356 跟踪

closed 2023-11-18 by wjz304