RROrg/rr #364: 关于手动指定网络配置IP

cnbb002, 2023-11-17:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU: i5-4590 
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  23.11.6
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS920+
 * version:  7.2

Issue:  

logs:  

(Plz review the content of #173, #175, #226 first)  
...
root@rr:/opt/rr# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether 00:23:24:86:b9:11 brd ff:ff:ff:ff:ff:ff
    inet 192.168.200.102/24 brd 192.168.200.255 scope global dynamic noprefixroute eth0
       valid_lft 7184sec preferred_lft 6284sec
    inet6 fe80::941c:5cf5:df74:781d/64 scope link
       valid_lft forever preferred_lft forever
root@rr:/opt/rr# cd /
root@rr:/# vi /etc/network/interfaces
root@rr:/# vi /etc/resolv.conf
root@rr:/# /etc/init.d/S40network restart
Stopping network: OK
Starting network: OK
root@rr:/# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether 00:23:24:86:b9:11 brd ff:ff:ff:ff:ff:ff
    inet 192.168.200.102/24 brd 192.168.200.255 scope global dynamic noprefixroute eth0
       valid_lft 6817sec preferred_lft 5917sec
    inet6 fe80::941c:5cf5:df74:781d/64 scope link
       valid_lft forever preferred_lft forever
root@rr:/# ping 114.114.114.114
PING 114.114.114.114 (114.114.114.114): 56 data bytes
^C
--- 114.114.114.114 ping statistics ---
5 packets transmitted, 0 packets received, 100% packet loss
root@rr:/# cat /etc/network/interfaces
# interface file auto-generated by buildroot

auto lo
iface lo inet loopback

iface eth0 inet static
address 128.5.254.4
netmask 255.255.255.0
gateway 128.5.254.254
root@rr:/# cat /etc/resolv.conf
# Generated by dhcpcd from eth0.dhcp
# /etc/resolv.conf.head can replace this line
nameserver 114.114.114.114
# /etc/resolv.conf.tail can replace this line
root@rr:/#

尝试修改interface和resolv.conf文件，填写正确ip地址和dns服务后，进行网络重启，命令执行OK后，但尝试用新的ip访问  发现并没有生效，恳请UP主指导一下，这种环境下怎么配置网络环境进行联网

Shuazijun, 2023-11-17:
[image]
你看这个，昨天作者说buildroot系统，那你直接搜这个就行了  
参考：[某WIKI](https://wiki.t-firefly.com/zh_CN/Firefly-Linux-Guide/manual_buildroot.html)

Shuazijun, 2023-11-17:
我这里测试是可以的，不过要先关闭DHCP自动获取IP，不然每次重启网络dns配置会被覆盖  
具体位置 /etc/init.d/S41dhcpcd   
cd进目录执行关闭指令 ./S41dhcpcd stop  
然后就是编辑 /etc/resolv.conf 如图  
[image]  
再编辑 /etc/network/interfaces 如图  
[image]  
随后直接进 etc/init.d/ 里面执行 ./S40network restart 就好了  
完成后再ping baidu就通了，
[image]

Shuazijun, 2023-11-17:
如图，已通网  
[image]

Shuazijun, 2023-11-17:
@wjz304 参考下看能不能做进脚本界面，现在的设置静态IP的选项没啥用，只对网络内有DHCP服务器的有效。

cnbb002, 2023-11-17:
> 如图，已通网 [image]

OK，按照提示操作，网络已联通。谢谢！准备入手后续的测试

wjz304, 2023-11-17:
UI 带式还能搞，关键是最好能保存 传递给dsm，不好搞哦

wjz304, 2023-11-18:
@Shuazijun @cnbb002  你俩 PR呗？

wjz304, 2023-11-18:
https://github.com/wjz304/rr/issues/358

Shuazijun, 2023-11-20:
@wjz304 =你研究，我们要是会，就不用啊arpl操作了

wjz304, 2024-06-27:
https://github.com/RROrg/rr/releases/tag/24.6.7

closed 2024-06-27 by wjz304