RROrg/rr #366: 能否排查一下SSD读写缓存盘重启后报错的问题？

asf5662004, 2023-11-18:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  VMware,
 * CPU:  AMD EPYC 7542 32-Core Processor
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  v23.11.6
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model: RS3621xs+
 * version:   7.2(69057u1)

Issue:  配置SSD读写缓存之后，重启DSM系统后，读写缓存报错。
[image]
[image]

配置：HDD、SSD都使用了RDM进行直通。

logs:  

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-11-18:
[image]

asf5662004, 2023-11-19:
> [image]

已经增加了日志

wjz304, 2023-11-20:
看不出来，无能为力，尝试hdddb插件 or 在磁盘管理里更新db数据