RROrg/rr #367: [Feature Request] 希望能添加virtio-balloon的内核驱动

hicasper, 2023-11-18:
ENV:
 * ENV:  Intel N100 + Proxmox VE QEMU KVM + iGPU SR-IOV
 * RR:  23.11.6
 * addons:  23.11.8
 * modules:  23.11.5
 * lkms:  23.10.0
DSM: 
 * model:  SA6400
 * version:  7.1.1-42962 Update 2

使用中发现能检测到控制器，但是没有virtio-balloon驱动，应该是群晖在编译的时候没有加入，希望能加入支持，毕竟每次虚拟机环境下要分配一大堆内存，但是又利用不起来怪可惜的。

`lspci -k -q` 的输出：
```
06:03.0 Unclassified device [00ff]: Red Hat, Inc. Virtio memory balloon
        Subsystem: Red Hat, Inc. Device 0005
        Kernel driver in use: virtio-pci
```

https://github.com/wjz304/arpl-zh_CN/blob/a8259099aefe926a6752e1c756da4da5b37a1492/files/board/arpl/kernel_defconfig#L432 此处发现有设置`CONFIG_VIRTIO_BALLOON=m`，但是在rr中没有找到相应的配置。

wjz304, 2023-11-18:
依赖太多，暂时没时间解决

wjz304, 2024-03-30:
SA6400 custom

closed 2024-03-30 by wjz304