RROrg/rr #372: 引导启动后丢失 /dev/synoboot 等4个设备文件

lighcen, 2023-11-21:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  [Acer Revo One (RL85)] 8G内存
 * CPU:  i3-4005U
 * NIC:  r8168 单网卡

RR version: (You can find it in the update menu.)  
 * RR:  23.11.6
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS918+
 * version:  7.2

Issue:  
在rr提示符下能够看到启动盘，图中sdb的8G启动盘。

成功进入安装界面，安装到 56% 时提示无法安装，这时telnet进去发现缺少 /dev/synoboot 等 4 个设备文件，fdisk -l 只看到数据盘 sda，没有启动盘 sdb.

是不是启动盘在引导后丢失导致不能安装文件损坏的提示呢？如何把 synoboot 设备文件找回来呢？

其他尝试
在其他电脑上安装好 7.2 再放到这台电脑上运行，可以成功启动，但ssh登陆后也是缺少 synoboot 这4个文件，在其他电脑安装时这些文件是存在的。洗白状态下 AME 激活打补丁成功，但最后返回“ame Patch is unsuccessful, retcode = 256”，怀疑也是与丢失synoboot有关。

[image]

logs:  

(Plz review the content of #173, #175, #226 first)  
...

xrh0905, 2023-11-23:
试试接到USB2.0上

lighcen, 2023-11-24:
> 试试接到USB2.0上

是连接USB2.0的，也换过U盘，同样引导后就看不到了。

wjz304, 2024-03-30:
请尝试最新版本并获取日志

closed 2024-03-30 by wjz304

hardbone12, 2024-07-24:
> > 试试接到USB2.0上
> 
> 是连接USB2.0的，也换过U盘，同样引导后就看不到了。

我和你的情况差不多，USB2.0接口的8G的启动盘在USB2.0上用不了，USB3.0上一切正常。
i7 2600K SandyBridge
RR: 24.7.3
DSM: DS3617xs-7.2.1-69057