RROrg/rr #374: 1265L v2 核显直通，群晖虚拟机关机后启动不了

alipig, 2023-11-21:
虚拟机 ： ESXI 7
ARPL 和 RR 都一样问题

虚拟机配置核显直通后启动完可以正常使用，只要群晖虚拟机一关机就启动不了，必须重启宿主机后才可以正常启动虚拟机

如果删除在虚拟机配置中删除直通的显卡是可以正常启动的，但是添加完直通显卡就不能启动

启动时的日志
异常信息1
2023-11-21T05:58:00.093Z Cr(01) vmx - PANIC: PCI passthru device 0000:00:02.0 caused an IOMMU fault type 5 at address 0x31bdc000.  Powering off the virtual machine.  If the problem persists please contact the device's vendor.
2023-11-21T05:58:00.898Z Wa(03) vmx - A core file is available in "/vmfs/volumes/5b2f9566-9f0b5e67-cc09-202564084fc4/synology/vmx-zdump.003"

异常信息2

2023-11-21T05:58:03.191Z In(05)+ vmx - PCI passthru device 0000:00:02.0 caused an IOMMU fault type 5 at address 0x31bdc000.  Powering off the virtual machine.  If the problem persists please contact the device's vendor.
2023-11-21T05:58:03.191Z In(05) vmx - [msg.panic.haveLog] A log file is available in "/vmfs/volumes/5b2f9566-9f0b5e67-cc09-202564084fc4/synology/vmware.log".  

附件（完整日志和dump文件）
[vmware.zip](https://github.com/wjz304/rr/files/13422379/vmware.zip)

alipig, 2023-11-21:
nstall ENV: (You can find it in the boot interface.)

DMI:
CPU: 1265L v2
NIC: 
RR version: (You can find it in the update menu.)

RR: 23.11.6 (RR和ARPL其他版本都一样)
addons: 
modules:
lkms:
DSM:

model:  DS920+
version: 7.2
Issue:
虚拟机配置核显直通后启动完可以正常使用，只要群晖虚拟机一关机就启动不了，必须重启宿主机后才可以正常启动虚拟机

如果删除在虚拟机配置中删除直通的显卡是可以正常启动的，但是添加完直通显卡就不能启动

logs:

异常信息1
2023-11-21T05:58:00.093Z Cr(01) vmx - PANIC: PCI passthru device 0000:00:02.0 caused an IOMMU fault type 5 at address 0x31bdc000. Powering off the virtual machine. If the problem persists please contact the device's vendor.
2023-11-21T05:58:00.898Z Wa(03) vmx - A core file is available in "/vmfs/volumes/5b2f9566-9f0b5e67-cc09-202564084fc4/synology/vmx-zdump.003"

异常信息2

2023-11-21T05:58:03.191Z In(05)+ vmx - PCI passthru device 0000:00:02.0 caused an IOMMU fault type 5 at address 0x31bdc000. Powering off the virtual machine. If the problem persists please contact the device's vendor.
2023-11-21T05:58:03.191Z In(05) vmx - [msg.panic.haveLog] A log file is available in "/vmfs/volumes/5b2f9566-9f0b5e67-cc09-202564084fc4/synology/vmware.log".

日志和dump文件附件
[vmware.zip](https://github.com/wjz304/rr/files/13422379/vmware.zip)

wjz304, 2023-11-21:
Edit the /etc/vmware/esx.conf file:
```
Change from: 
/device/000:02.0/owner = "passthru" 
Change to: 
/device/000:02.0/owner = "vmkernel"
```

wjz304, 2023-11-21:
https://kb.vmware.com/s/article/1022011

alipig, 2023-11-22:
> Edit the /etc/vmware/esx.conf file:
> 
> ```
> Change from: 
> /device/000:02.0/owner = "passthru" 
> Change to: 
> /device/000:02.0/owner = "vmkernel"
> ```

这是关闭直通的，没用。我还以为引导的问题。估计是虚拟机关机了，直通设备占用的内存没释放

wjz304, 2023-11-24:
我不用esxi，不是很了解，看有没有其他人吧

wjz304, 2024-03-30:
复现请重新打开

closed 2024-03-30 by wjz304