RROrg/rr #377: Support connexion with proxy

niko44, 2023-11-22:
Hi,

Thks for your work. 
Is it possible to add support to configure an https proxy in a submenu ?

niko44, 2023-11-22:
When the loader need to get the pat file, it is mandatory to reach it from the Internet but some times it is not possible to do it without exit from a proxy, for example in a company.

I tried exit the menu and configure 2 variables environnement  (HTTP_PROXY and HTTPS_PROXY) but it seems that the client http who check the pat file don't use it. So I can't install the loader in my company :/

niko44, 2023-11-22:
I saw that the loader use curl to get pat file. So the -x o --proxy param is need like the man curl :

`-x, --proxy <[protocol://][user:password@]proxyhost[:port]>

     Use the specified HTTP proxy. 
     If the port number is not specified, it is assumed at port 1080.`

niko44, 2023-11-22:
I saw that the loader use curl to get pat file. So the -x or --proxy param is need like the man curl :

`-x, --proxy <[protocol://][user:password@]proxyhost[:port]>

     Use the specified HTTP proxy. 
     If the port number is not specified, it is assumed at port 1080.`

wjz304, 2023-11-22:
I probably understand, 
On the other hand, some people in China actually add agents mainly for updates.

I will reconsider adding it based on the two needs

niko44, 2023-11-22:
Finally I did the trick on the host with environnement variable 'export https_proxy=http://XX.XX.XX.XX:8080/' for example before lunch menu.sh. I don't know why is such long but it's works!

wjz304, 2023-11-22:
其实一直有效，这是linux 通用了

niko44, 2023-11-22:
yes it's native to Linux but it's not super user friendly!
This requires:

1.   to change the keyboard keymap
2.   to quit the menu
3.   to enter the export command https_proxy="http://IP:PORT"
4.   to restart the menu.sh
and all from the host and not remotely.

wjz304, 2023-11-26:
https://github.com/wjz304/rr/releases/tag/23.11.9

wjz304, 2023-11-26:
I forgot to write this in the release log,

wjz304, 2023-11-26:
[image]

niko44, 2023-11-28:
thank you very much ;)

closed 2023-11-28 by niko44