RROrg/rr #378: Since version rr 11.5, the LAN card is not detected after rebooting in the 918+ model.

Gurabity, 2023-11-22:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  g4560
 * NIC:  r8168

RR version: (You can find it in the update menu.)  
 * RR:  11.4
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  ds918+
 * version:  7.2

Issue:  Starting with version rr 11.5, on 918+ models the LAN card is not detected after reboot.
It works correctly in version 11.4.
In version 11.5, you can connect for the first time after updating and rebuilding, but you cannot connect after rebooting.

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

id10000, 2023-11-22:
me too,920+, 7.1

wjz304, 2023-11-22:
```
lspci -Qnnk
dmesg | grep r8168
lsmod
```

wjz304, 2024-03-30:
复现请重新打开

closed 2024-03-30 by wjz304