RROrg/rr #380: DSM异常关机

nmbvd, 2023-11-24:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  QENU standrad PC (q35+ICH9,2009)
 * CPU:  QEMU Virtual CPU version 2.5+ （物理CPU i5-4590)
 * NIC:  virtio （物理网卡Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 ）

RR version: (You can find it in the update menu.)  
 * RR:  23.11.8
 * addons:  23.11.7
 * modules:  23.11.10
 * lkms:  23.10.0

DSM: 
 * model:  DS920+
 * version:  7.2.1-69057

Issue:  在物理主机装proxmox, 然后在虚拟机里安装DSM，运行1-2个小时DSM提示异常关机

logs:  
重启时在群晖里的报错
[image]
[image]
群晖/var/log/messages里的启动报错
[log.txt](https://github.com/wjz304/rr/files/13457695/log.txt)

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

Promix953, 2023-11-24:
不是带ERROR字样的就是报错，建议放完整日志文件上来

顺便问下：有在PT刷流之类的么？

nmbvd, 2023-11-24:
重启了2次以后就没再重启了 ，没有用PT下载。取log的话按照 #173 里的是不是要重建引导？原来的系统就没了吧，这样影响好大

Promix953, 2023-11-24:
> 重启了2次以后就没再重启了 ，没有用PT下载。取log的话按照 #173 里的是不是要重建引导？原来的系统就没了吧，这样影响好大

DSM系统里的/var/log/kern.log和完整的/var/log/messages

nmbvd, 2023-11-24:
>

> > 重启了2次以后就没再重启了 ，没有用PT下载。取log的话按照 #173 里的是不是要重建引导？原来的系统就没了吧，这样影响好大
> 
> DSM系统里的/var/log/kern.log和完整的/var/log/messages

log如下
[kern.log](https://github.com/wjz304/rr/files/13460637/kern.log)
[messages.zip](https://github.com/wjz304/rr/files/13460657/messages.zip)

mrbaiwei, 2023-11-26:
[image]
我遇到的异常重启是某个docker服务(CPU相对比较高)造成，我停止后就不再出现

wjz304, 2024-03-30:
复现请重新打开

closed 2024-03-30 by wjz304