RROrg/rr #381: BUG

LD166, 2023-11-24:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  N100
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  11.8
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400 和 DS3662
 * version:  7.2.1

Issue:  定时开机问题

logs:  
以下是详细情况：
N100 CPU 安装SA3622 最新7.2.1系统，直接在系统电源配置页面选择时间就可以开关机了，但是SA6400 最新7.2.1系统同样操作就是不行，开启SA6400 和 DS3622的 SSH 查看rtc 信息如下图所示：
[image]
[image]
3622对比6400多了一个HPET_emulated ，感觉开机和HPET_emulated有关，发出来大佬们研究下。

wjz304, 2023-11-24:
方便的话 你试一下 918/920 看看是否好使，

LD166, 2023-11-24:
> 方便的话 你试一下 918/920 看看是否好使，

试过了918也是开不了机，同时 **HPET_emulated ：no**
[image]

LD166, 2023-11-24:
> 代码逻辑看只有 ds3622/dva3221 此项是开启的

那应该就是这个dva3221 的问题，我下午查资料有个文章也是说的这个dva3221 开启

wjz304, 2023-11-24:
我编个版本 你试试，1个小时吧

LD166, 2023-11-24:
OK 。怎么发我?

wjz304, 2023-11-24:
看简单了，依赖太多，编不了

LD166, 2023-11-24:
> 看简单了，依赖太多，编不了
哈哈 下次版本能更新么？

wjz304, 2023-11-24:
> > 看简单了，依赖太多，编不了
> > 哈哈 下次版本能更新么？

依赖解决不了

LD166, 2023-11-24:
> > > 看简单了，依赖太多，编不了
> > > 哈哈 下次版本能更新么？
> 
> 依赖解决不了
好的

wjz304, 2023-11-24:
不像网卡驱动，这些涉及内核参数的东西很困难，

LD166, 2023-11-24:
嗯嗯 谢谢大佬

Promix953, 2023-11-25:
@LD166 你有没有安装`powersched`这个插件？

LD166, 2023-11-25:
> powersched

哪里的？矿？

Promix953, 2023-11-25:
> > powersched 电源调度
> 
> 哪里的？矿？

。。。
RR菜单里面，中文->插件， 英文->Addons

LD166, 2023-11-25:
> 文->Addons

没试过，你试过么？有用？

Promix953, 2023-11-25:
> > 文->Addons
> 
> 没试过，你试过么？有用？

我用不到这个功能，你可以试试行不行，插件的功能介绍是：
`Utility to configure RTC according to DSM power schedule`
翻译过来是：
`根据 DSM 电源计划配置 RTC 的实用程序`

LD166, 2023-11-25:
> > > 文->Addons
> > 
> > 
> > 没试过，你试过么？有用？
> 
> 我用不到这个功能，你可以试试行不行，插件的功能介绍是： `Utility to configure RTC according to DSM power schedule` 翻译过来是： `根据 DSM 电源计划配置 RTC 的实用程序`
哦哦 稍等我试下

LD166, 2023-11-25:
> > > 文->Addons
> > 
> > 
> > 没试过，你试过么？有用？
> 
> 我用不到这个功能，你可以试试行不行，插件的功能介绍是： `Utility to configure RTC according to DSM power schedule` 翻译过来是： `根据 DSM 电源计划配置 RTC 的实用程序`

牛啊。确实可以很强，这个插件支持硬盘休眠么？

Promix953, 2023-11-25:
> > > > 文->Addons
> > > 
> > > 
> > > 没试过，你试过么？有用？
> > 
> > 
> > 我用不到这个功能，你可以试试行不行，插件的功能介绍是： `Utility to configure RTC according to DSM power schedule` 翻译过来是： `根据 DSM 电源计划配置 RTC 的实用程序`
> 
> 牛啊。确实可以很强，这个插件支持硬盘休眠么？

硬盘休眠看另一个issue吧，那边有写 #382

wjz304, 2023-11-25:
powersched

closed 2023-11-25 by wjz304