RROrg/rr #382: Unable to put NAS into hibernation

AdictiK5, 2023-11-24:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  Q2900M - J2900
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  23.11.8
 * addons:  23.11.7
 * modules:  23.11.10
 * lkms:  23.10.0

DSM: 
 * model:  DS2422+
 * version:  7.2(69057u2)

Issue:  Unable to put NAS into hibernation

I previously had problems with hybernation when running under ARPL. 
The problems were corrected with fbelavenuto (https://github.com/fbelavenuto/arpl/issues/309). 
Since I migrated to "RR" my problems have returned. 
Can you help me? If you tell me what to do I can provide you with the missing information. 
Thanks in advance.

Promix953, 2023-11-25:
Did you modify `/etc.defaults/syslog-ng/patterndb.d/scemd.conf` and `/etc/syslog-ng/patterndb.d/synosystemd.conf`?
Can you send the result of the following command:
```
systemctl list-units --state failed
dmesg | tail
tail /var/log/kern.log
tail /var/log/message
```

AdictiK5, 2023-11-25:
No I didn't touch these two files.

AdictiK@XServerNAS:~$ systemctl list-units --state failed
  UNIT                         LOAD   ACTIVE SUB    DESCRIPTION
● pkg-LogCenter-recvrule.service loaded failed failed
 Log Center syslog server
● pkgctl-HyperBackup-ED.service loaded failed failed 
HyperBackup-ED's service uni
● synoindex-checkpackage.service loaded failed failed
 synoindex check if there are
● SynoInitEth.service          loaded failed failed A
djust NIC sequence

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

4 loaded units listed. Pass --all to see loaded but inactive units, too
.
To show all installed unit files use 'systemctl list-unit-files'.
AdictiK@XServerNAS:~$ dmesg | tail
[25312.664209] Cannot get i2c adapter
[28918.459440] Cannot get i2c adapter
[32524.245856] Cannot get i2c adapter
[36130.086704] Cannot get i2c adapter
[39736.283611] Cannot get i2c adapter
[43342.023668] Cannot get i2c adapter
[46947.848449] Cannot get i2c adapter
[50553.792210] Cannot get i2c adapter
[54159.494902] Cannot get i2c adapter
[57765.774797] Cannot get i2c adapter
AdictiK@XServerNAS:~$ tail /var/log/kern.log
tail: cannot open '/var/log/kern.log' for reading: Permission denied
AdictiK@XServerNAS:~$ tail /var/log/message
tail: cannot open '/var/log/message' for reading: No such file or directory
AdictiK@XServerNAS:~$

Promix953, 2023-11-25:
> No I didn't touch these two files.不，我没有碰这两个文件。
> 
> AdictiK@XServerNAS:~$ systemctl list-units --state failedAdictiK@XServerNAS:~$ systemctl list-units --state 失败 UNIT LOAD ACTIVE SUB DESCRIPTION 单元负载活动子描述 ● pkg-LogCenter-recvrule.service loaded failed failed● pkg-LogCenter-recvrule.service 加载失败失败 Log Center syslog server日志中心系统日志服务器 ● pkgctl-HyperBackup-ED.service loaded failed failed ● pkgctl-HyperBackup-ED.service 加载失败失败 HyperBackup-ED's service uni HyperBackup-ED的服务单元 ● synoindex-checkpackage.service loaded failed failed ● synoindex-checkpackage.service 加载失败 失败 synoindex check if there aresynoindex 检查是否有 ● SynoInitEth.service loaded failed failed A● SynoInitEth.service加载失败失败A djust NIC sequence 调整网卡顺序
> 
> LOAD = Reflects whether the unit definition was properly loaded.LOAD = 反映单元定义是否已正确加载。 ACTIVE = The high-level unit activation state, i.e. generalization of SUB. ACTIVE = 高级单元激活状态，即 SUB 的泛化。 SUB = The low-level unit activation state, values depend on unit type. SUB = 低级单元激活状态，值取决于单元类型。
> 
> 4 loaded units listed. Pass --all to see loaded but inactive units, too列出 4 个加载单元。通过 --all 也可以查看已加载但不活动的单元 . To show all installed unit files use 'systemctl list-unit-files'. 要显示所有已安装的单元文件，请使用“systemctl list-unit-files”。 AdictiK@XServerNAS:~$ dmesg | tail [25312.664209] Cannot get i2c adapter[25312.664209] 无法获取 i2c 适配器 [28918.459440] Cannot get i2c adapter [28918.459440] 无法获取 i2c 适配器 [32524.245856] Cannot get i2c adapter [32524.245856] 无法获取 i2c 适配器 [36130.086704] Cannot get i2c adapter [36130.086704] 无法获取 i2c 适配器 [39736.283611] Cannot get i2c adapter [39736.283611] 无法获取 i2c 适配器 [43342.023668] Cannot get i2c adapter [43342.023668] 无法获取 i2c 适配器 [46947.848449] Cannot get i2c adapter [46947.848449] 无法获取 i2c 适配器 [50553.792210] Cannot get i2c adapter [50553.792210] 无法获取 i2c 适配器 [54159.494902] Cannot get i2c adapter [54159.494902] 无法获取 i2c 适配器 [57765.774797] Cannot get i2c adapter [57765.774797] 无法获取 i2c 适配器 AdictiK@XServerNAS:~AdictiK@XServerNAS:$ dmesg |尾巴$ tail /var/log/kern.logAddictiK@XServerNAS:$ tail /var/log/kern.log tail: cannot open '/var/log/kern.log' for reading: Permission deniedtail：无法打开“/var/log/kern.log”进行读取：权限被拒绝 AdictiK@XServerNAS:~$ tail /var/log/message tail: cannot open '/var/log/message' for reading: No such file or directorytail：无法打开“/var/log/message”进行读取：没有这样的文件或目录 AdictiK@XServerNAS:~$ 上瘾者@XServerNAS:$

You need to run these commands with root privileges -> `sudo -i`
### RUN AS ROOT！！！
## RUN AS ROOT！！！
# RUN AS ROOT！！！

``` shell
   sed 's|^destination d_scemd { file("/dev/null"); };|destination d_scemd { file("/var/log/scemd.log"); };|' -i /tmpRoot/etc.defaults/syslog-ng/patterndb.d/scemd.conf
   sed 's|^destination d_scemd { file("/dev/null"); };|destination d_scemd { file("/var/log/scemd.log"); };|' -i /tmpRoot/etc/syslog-ng/patterndb.d/scemd.conf
   sed 's|^destination d_synosystemd { file("/dev/null"); };|destination d_synosystemd { file("/var/log/synosystemd.log"); };|' -i /tmpRoot/etc.defaults/syslog-ng/patterndb.d/synosystemd.conf
   sed 's|^destination d_synosystemd { file("/dev/null"); };|destination d_synosystemd { file("/var/log/synosystemd.log"); };|' -i /tmpRoot/etc/syslog-ng/patterndb.d/synosystemd.conf
   sed 's|^destination d_systemd { file("/dev/null"); };|destination d_systemd { file("/var/log/synosystemd.log"); };|' -i /tmpRoot/etc.defaults/syslog-ng/patterndb.d/synosystemd.conf
   sed 's|^destination d_systemd { file("/dev/null"); };|destination d_systemd { file("/var/log/synosystemd.log"); };|' -i /tmpRoot/etc/syslog-ng/patterndb.d/synosystemd.conf
   sed 's|flags(final);||' -i /tmpRoot/etc.defaults/syslog-ng/patterndb.d/scemd.conf
   sed 's|flags(final);||' -i /tmpRoot/etc/syslog-ng/patterndb.d/scemd.conf
   sed 's|flags(final);||' -i /tmpRoot/etc.defaults/syslog-ng/patterndb.d/synosystemd.conf
   sed 's|flags(final);||' -i /tmpRoot/etc/syslog-ng/patterndb.d/synosystemd.conf
```

AdictiK5, 2023-11-25:
Ok I ran the previous commands in admin

root@XServerNAS:~# systemctl list-units --state failed
  UNIT                         LOAD   ACTIVE SUB    DESCRIPTION
● pkg-LogCenter-recvrule.service loaded failed failed
 Log Center syslog server
● pkgctl-HyperBackup-ED.service loaded failed failed 
HyperBackup-ED's service uni
● synoindex-checkpackage.service loaded failed failed
 synoindex check if there are
● SynoInitEth.service          loaded failed failed A
djust NIC sequence

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

4 loaded units listed. Pass --all to see loaded but inactive units, too
.
To show all installed unit files use 'systemctl list-unit-files'.
root@XServerNAS:~# dmesg | tail
[61405.589737] audit: type=1325 audit(1700912881.193:109): table=filter family=2 entries=14
[61405.597696] audit: type=1325 audit(1700912881.201:110): table=filter family=2 entries=14
[61405.605167] audit: type=1325 audit(1700912881.209:111): table=filter family=10 entries=26
[61405.612593] audit: type=1325 audit(1700912881.216:112): table=filter family=10 entries=26
[61405.620581] audit: type=1325 audit(1700912881.224:113): table=filter family=10 entries=26
[61405.628319] audit: type=1325 audit(1700912881.232:114): table=filter family=2 entries=18
[61405.636107] audit: type=1325 audit(1700912881.240:115): table=filter family=10 entries=30
[61405.643759] audit: type=1325 audit(1700912881.247:116): table=filter family=2 entries=13
[61405.651442] audit: type=1325 audit(1700912881.255:117): table=filter family=10 entries=19
[64978.694917] Cannot get i2c adapter
root@XServerNAS:~# tail /var/log/kern.log
2023-11-25T04:47:26+01:00 XServerNAS kernel: [32524.245856] Cannot get i2c adapter
2023-11-25T05:47:26+01:00 XServerNAS kernel: [36130.086704] Cannot get i2c adapter
2023-11-25T06:47:26+01:00 XServerNAS kernel: [39736.283611] Cannot get i2c adapter
2023-11-25T07:47:26+01:00 XServerNAS kernel: [43342.023668] Cannot get i2c adapter
2023-11-25T08:47:26+01:00 XServerNAS kernel: [46947.848449] Cannot get i2c adapter
2023-11-25T09:47:26+01:00 XServerNAS kernel: [50553.792210] Cannot get i2c adapter
2023-11-25T10:47:26+01:00 XServerNAS kernel: [54159.494902] Cannot get i2c adapter
2023-11-25T11:47:27+01:00 XServerNAS kernel: [57765.774797] Cannot get i2c adapter
2023-11-25T12:47:28+01:00 XServerNAS kernel: [61372.659927] Cannot get i2c adapter
2023-11-25T13:47:28+01:00 XServerNAS kernel: [64978.694917] Cannot get i2c adapter
root@XServerNAS:~# tail /var/log/message
tail: cannot open '/var/log/message' for reading: No such file or directory

------------

I just executed the other 3 commands. And I rebooted

Promix953, 2023-11-25:
@AdictiK5
Okay, the last 3 commands are what arpl did.
I think you could try to hibernate the disk.

wjz304, 2023-11-25:
> @AdictiK5 Okay, the last 3 commands are what arpl did. I think you could try to hibernate the disk.

[image]
格式不一致吧

wjz304, 2023-11-25:
加个addon 吧

AdictiK5, 2023-11-25:
> 加个addon 吧

Which one should I choose?

AdictiK5, 2023-11-25:
> @AdictiK5 Okay, the last 3 commands are what arpl did. I think you could try to hibernate the disk.

I did a reboot, still no hybernation at the moment.

Promix953, 2023-11-25:
> > @AdictiK5 Okay, the last 3 commands are what arpl did. I think you could try to hibernate the disk.好的，最后 3 个命令就是 arpl 所做的。我想你可以尝试让磁盘休眠。
> 
> I did a reboot, still no hybernation at the moment.我重新启动了，目前仍然没有休眠。

Try these：
``` shell
sudo -i
systemctl stop pkg-LogCenter-recvrule.service
systemctl stop pkgctl-HyperBackup-ED.service
systemctl stop synoindex-checkpackage.service
systemctl stop SynoInitEth.service
systemctl mask pkg-LogCenter-recvrule.service
systemctl mask pkgctl-HyperBackup-ED.service
systemctl mask synoindex-checkpackage.service
systemctl mask SynoInitEth.service
```

AdictiK5, 2023-11-25:
> Try these：
>
> ```shell
> sudo -i
> systemctl stop pkg-LogCenter-recvrule.service
> systemctl stop pkgctl-HyperBackup-ED.service
> systemctl stop synoindex-checkpackage.service
> systemctl stop [...]

-------

Ok, I ran the commands, I'm going to reboot.

root@XServerNAS:~# sudo -i
root@XServerNAS:~# systemctl stop pkg-LogCenter-recvrule.service
root@XServerNAS:~# systemctl stop pkgctl-HyperBackup-ED.service
root@XServerNAS:~# systemctl stop synoindex-checkpackage.service
root@XServerNAS:~# systemctl stop SynoInitEth.service
root@XServerNAS:~# systemctl mask pkg-LogCenter-recvrule.service
Created symlink from /etc/systemd/system/pkg-LogCenter-recvrule.service to /dev/null.
root@XServerNAS:~# systemctl mask pkgctl-HyperBackup-ED.service
Created symlink from /etc/systemd/system/pkgctl-HyperBackup-ED.service to /dev/null.
root@XServerNAS:~# systemctl mask synoindex-checkpackage.service
Created symlink from /etc/systemd/system/synoindex-checkpackage.service to /dev/null.
root@XServerNAS:~# systemctl mask SynoInitEth.service

Promix953, 2023-11-25:
> 加个addon 吧

scemd是必加的
systemd我是折腾完没多余日志了（所以修了acpid😂），但是每个机型都会不同，只屏蔽掉synosystemd.log和systemd.log的话，还是会有systemd/*.service.log，需要针对性mask掉service

LD166, 2023-11-25:
> > 加个addon吧
> 
> scemd是必加的 systemd我是折腾完没秃日志了（所以修改了acpid😂），但是每个机舱都会不同，只消掉synosystemd.log和systemd.log的话，还是会有systemd/*.service。 log，需要屏蔽掉服务
所以说是 log 导致了不能休眠？

Promix953, 2023-11-25:
@AdictiK5 
```
root@XServerNAS:~# systemctl mask SynoInitEth.service
```
⬆️⬆️⬆️Don't forget to press ENTER⬆️⬆️⬆️

Promix953, 2023-11-25:
> > > 加个addon吧
> > 
> > 
> > scemd是必加的 systemd我是折腾完没秃日志了（所以修改了acpid😂），但是每个机舱都会不同，只消掉synosystemd.log和systemd.log的话，还是会有systemd/*.service。 log，需要屏蔽掉服务
> > 所以说是 log 导致了不能休眠？

1. 日志默认是写在系统分区的，系统分区又是所有盘（还是前12盘来着？记不清了）的raid1，只要持续写日志，硬盘就不可能休眠
2. 解决了日志问题还有套件问题，这个是黑白群共通的，官方有相关说明，例如docker就影响硬盘休眠

LD166, 2023-11-25:
> > > > 加个addon吧
> > > 
> > > 
> > > scemd是必加的 systemd我是折腾完没秃日志了（所以修改了acpid😂），但是每个机舱都会不同，只消掉synosystemd.log和systemd.log的话，还是会有systemd/*.service。 log，需要屏蔽掉服务
> > > 所以说是 log 导致了不能休眠？
> 
> 1. 日志默认是写在系统分区的，系统分区又是所有盘（还是前12盘来着？记不清了）的raid1，只要持续写日志，硬盘就不可能休眠
> 2. 解决了日志问题还有套件问题，这个是黑白群共通的，官方有相关说明，例如docker就影响硬盘休眠

哦哦 我的辣鸡白群就让他存重要的东西，然后一直休眠。黑白套件问题肯定都是一样的。

wjz304, 2023-11-25:
> > 加个addon 吧
> 
> scemd是必加的 systemd我是折腾完没多余日志了（所以修了acpid😂），但是每个机型都会不同，只屏蔽掉synosystemd.log和systemd.log的话，还是会有systemd/*.service.log，需要针对性mask掉service

https://github.com/wjz304/rr-addons/blob/main/hibernation/install.sh

服务不好搞啊... 不好还原

AdictiK5, 2023-11-25:
> @AdictiK5
> 
> ```
> root@XServerNAS:~# systemctl mask SynoInitEth.service
> ```
> 
> ⬆️⬆️⬆️Don't forget to press ENTER⬆️⬆️⬆️

OK i try again, thanks

Promix953, 2023-11-25:
> > > 加个addon 吧
> > 
> > 
> > scemd是必加的 systemd我是折腾完没多余日志了（所以修了acpid😂），但是每个机型都会不同，只屏蔽掉synosystemd.log和systemd.log的话，还是会有systemd/*.service.log，需要针对性mask掉service
> 
> https://github.com/wjz304/rr-addons/blob/main/hibernation/install.sh
> 
> 服务不好搞啊... 不好还原

还原倒是挺简单
mask <-> unmask
但是除了固定几个必失败的系统服务，还有pkg的服务也有这个情况，比如：`pkg-SynologyPhotos-js-server.service`

Promix953, 2023-11-25:
> > > > 加个addon 吧
> > > 
> > > 
> > > scemd是必加的 systemd我是折腾完没多余日志了（所以修了acpid😂），但是每个机型都会不同，只屏蔽掉synosystemd.log和systemd.log的话，还是会有systemd/*.service.log，需要针对性mask掉service
> > 
> > 
> > https://github.com/wjz304/rr-addons/blob/main/hibernation/install.sh
> > 服务不好搞啊... 不好还原
> 
> 还原倒是挺简单 mask <-> unmask 掩码 <-> 取消掩码 但是除了固定几个必失败的系统服务，还有pkg的服务也有这个情况，比如：`pkg-SynologyPhotos-js-server.service`

最简单粗暴有效的方式其实是把/var/log扔Ramdisk。。。

wjz304, 2023-11-25:
> 还原倒是挺简单 mask <-> unmask 但是除了固定几个必失败的系统服务，还有pkg的服务也有这个情况，比如：`pkg-SynologyPhotos-js-server.service`

如何不同型号不同机器不一样 就要记住关闭了那些服务，原来是什么状态

Promix953, 2023-11-25:
> 如何不同型号不同机器不一样 就要记住关闭了那些服务，原来是什么状态

是的，还涉及到系统升级、套件升级可能的变化，要疯😂😂😂
哎，当爹当妈总有要放手的那一天，有些活还是要自己动手最直接

Promix953, 2023-11-25:
> 如何不同型号不同机器不一样 就要记住关闭了那些服务，原来是什么状态

关键全解决了还不一定能硬盘休眠哦
[https://kb.synology.cn/zh-cn/DSM/tutorial/What_stops_my_Synology_NAS_from_entering_System_Hibernation](url)
所以我是直接关了硬盘休眠的

Promix953, 2023-11-25:
> 如何不同型号不同机器不一样 就要记住关闭了那些服务，原来是什么状态

我想到一个方案
scemd没什么好说的直接flags(final)
systemd这边不用flags(final)，改为建个开机schedule，延迟一个时间，比如60s，获取失败的service，stop掉
这样侵入性最小

AdictiK5, 2023-11-25:
Still no hybernation unfortunately :( 
I need to activate an additional addon? 
I need to deactivate a function in the system?

Promix953, 2023-11-26:
> Still no hybernation unfortunately :( 
> I need to activate an additional addon? 
> I need to deactivate a function in the system?

You need to make two scripts - on boot-up and shutdown. It is very easy in DSM interface using its control panel. Both scripts run as root and has any names you like.
On boot-up:
```
if [ ! -d "/var/logbak" ]; then
mkdir /var/logbak
cp -a -f /var/log/. /var/logbak/
fi
cp -a -f /var/logbak/. /tmp/log/
mount -B /tmp/log /var/log
if [ ! -d "/var/cachebak" ]; then
mkdir /var/cachebak
cp -a -f /var/cache/samba/. /var/cachebak/
fi
cp -a -f /var/cachebak/. /tmp/cache/
mount -B /tmp/cache /var/cache/samba
```

On shutdown:
```
if [ ! -d "/var/logbak" ]; then
cp -a -f /tmp/log/. /var/logbak
fi
if [ ! -d "/var/cachebak" ]; then
cp -a -f /tmp/cache/. /var/cachebak
fi
```
And then reboot. Everything should now work - both samba and hibernation. As we store all logs in ram - there is no practical need to do anything with them. But I think that it is better to leave as done now, as this two logs contain only garbage with ARPL. On the other hand I do not know how to examine boot logs that were produced before first script (but dmesg works). And as for ARPL - there may be some more elegant solution, I don't know...
And the above scripts can be reduced twice - if you abondon first half about /var/log. And hibernation will work. But hdd-s will wake up if somebody will write to logs. It happens several times a day.
Also I am not sure that samba works with previous session cache tdb-s. If not - scripts may be reduced on writing its back on shutdown.

AdictiK5, 2023-11-26:
I have set up both scripts, waiting to see if it works ;)
It's a shame to have something that worked with ARPL and following the update to RR it no longer works :(

Promix953, 2023-11-26:
> I have set up both scripts, waiting to see if it works ;)
> It's a shame to have something that worked with ARPL and following the update to RR it no longer works :(

There are many differences between each dsm version, each model, and different kits installed by each person. There are many hard disk hibernation issues even with real synology devices. There is no absolute universal method. But there is a lot of experience and tutorials on the web.

If hard disk hibernation is very important to you, I'd suggest that it's better to fall back to ARPL and enjoying it.

wjz304, 2025-05-08:
https://github.com/RROrg/rr-tools#:~:text=Custom%20blocking%20logs%20for%20Synology%20system