RROrg/rr #384: 添加硬盘，计算RAID无限重启

dreamfly-jiefa, 2023-11-25:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

dreamfly-jiefa, 2023-11-25:
在说明一个情况，如果黑群安装完之后，简单添加base磁盘，或者组两块SHR，（即不需要计算磁盘数据），整体都会运行很稳定，没有任何问题。只要涉及大量磁盘计算，必死

wjz304, 2023-11-25:
[image]
磁盘管理界面有个更新db数据，更新下试试

Promix953, 2023-11-25:
从报错文件字面意思上看是存储管理器的配置文件有问题？
我这`/var/lib/storage_setting/`下面是空的。。。
把`/var/lib/storage_setting/general_settings.db`发出来看下？

wjz304, 2023-11-25:
[image]
[image]
https://www.synology.cn/zh-cn/support/download/DS920+?version=7.2#system

dreamfly-jiefa, 2023-11-25:
> 从报错文件字面意思上看是存储管理器的配置文件有问题？ 我这`/var/lib/storage_setting/`下面是空的。。。 把`/var/lib/storage_setting/general_settings.db`发出来看下？

[image]
确实是有这个文件，但是文件内容0kb，空的

dreamfly-jiefa, 2023-11-25:
> ![image](https://private-user-images.githubusercontent.com/5615843/285568620-79c2c4d6-3c54-45ea-ad25-adf1d325e1ee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA4OTI5NTksIm5iZiI6MTcwMDg5MjY1OSwicGF0aCI6Ii81NjE1ODQzLzI4NTU2ODYyMC03OWMyYzRkNi0zYzU0LTQ1ZWEtYWQyNS1hZGYxZDMyNWUxZWUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTEyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzExMjVUMDYxMDU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTg3MTY2YTNhODZmOTE1ZDA3NmIxODI3ZTZmOGNjOGYyMGFlN2UxYjNlODZlMzZiMDE2ZTNhN2I3OGI4NTUzZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.qq5Rebci3GEyFNnCbAcPNLNGhpjHitHZ546VzYu77r4) ![image](https://private-user-images.githubusercontent.com/5615843/285568659-8a83e314-80a6-4dd6-a2c9-fe384b689135.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDA4OTI5NTksIm5iZiI6MTcwMDg5MjY1OSwicGF0aCI6Ii81NjE1ODQzLzI4NTU2ODY1OS04YTgzZTMxNC04MGE2LTRkZDYtYTJjOS1mZTM4NGI2ODkxMzUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTEyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzExMjVUMDYxMDU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmRkOWU0ZGU2ZmVjZmMyNWIwZjNmOGU3OTFiZDE0MTllYTIxZDkwNmY2NWVmZjgyNWY5ZjkyYWY4MGU3NDRkYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Rf3dj776KGDS2ybTdFBzoVjWpmj6JBtDKGvf-z6pxfs) https://www.synology.cn/zh-cn/support/download/DS920+?version=7.2#system

[image]
官方下载了最新版本，结果提示版本过低。又下了一次，一样的提示

wjz304, 2023-11-25:
型号没错的话，等有时间我装一个 提取给你吧

dreamfly-jiefa, 2023-11-25:
> 型号没错的话，等有时间我装一个 提取给你吧

好的，谢谢，我转给的是AS6400,7.2版本

wjz304, 2023-11-27:
[storage_setting.zip](https://github.com/wjz304/rr/files/13469939/storage_setting.zip)
[image]

wjz304, 2024-03-30:
复现请重新打开

closed 2024-03-30 by wjz304