RROrg/rr #385: Even in version 23.11.8, the LAN card is not recognized after rebooting in the 918+ model.

Gurabity, 2023-11-25:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  g4560
 * NIC:  r8168

RR version: (You can find it in the update menu.)  
 * RR:  23.11.8
 * addons:  
 * modules:  all
 * lkms:  

DSM: 
 * model:  ds918+
 * version:  7.2

Issue:  
LAN does not work after reboot as expected in version 11.8.
It works correctly in version 11.4.
In the end, we rolled back to version 11.4.
Please act quickly...

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2023-11-25:
RR? Or DSM?

Gurabity, 2023-11-25:
Update rr to the latest version and proceed with rebuild.
Afterwards, the first DSM is installed properly and operates normally.
However, if I restart DSM, DSM will not go into it afterward.
If you check in rr, after the first boot, it reboots into kernel mode and then the LAN dies.

wjz304, 2023-11-25:
> Afterwards, the first DSM is installed properly and operates normally.

My understanding is that after the upgrade, the system can be entered normally and the LAN is available, right?

> However, if I restart DSM, DSM will not go into it afterward.
> If you check in rr, after the first boot, it reboots into kernel mode and then the LAN dies.

Then I restarted, and it didn't start normally, but got stuck in "kernel mode"?

Can you provide a picture of "kernel mode"?

If you enter DSM system normally for the first time and the lan is OK, then there is no problem with the network card driver.
I think the "kernel mode" you're referring to is kernel panic.

Gurabity, 2023-11-26:
[logs.zip](https://github.com/wjz304/rr/files/13466185/logs.zip)

After updating again, I checked the symptoms.
It boots normally, and the connection to DSM is confirmed on the screen. When checked with the router, the LAN comes up normally, but it does not connect, and the LAN dies within 5 minutes.
I don't know what the logs are, so I'm attaching them all.

wjz304, 2023-11-26:
log
```
[    6.935145] r8169_lk Gigabit Ethernet driver 2.3LK-NAPI loaded
[    6.935149] r8169_lk 0000:01:00.0: can't disable ASPM; OS doesn't have ASPM control
...
[   25.467389] r8169_lk 0000:01:00.0 eth0: link down
[   25.467389] r8169_lk 0000:01:00.0 eth0: link down
[   25.477150] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   28.476076] r8169_lk 0000:01:00.0 eth0: link up
[   28.480761] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

...
01:00.0 Class [0200]: Device [10ec:8168] (rev 15)
	Subsystem: Device [1019:8117]
```

1. There is an ASPM warning, but ASPM was closed in 23.11.7. You said that it will not work from 23.11.5, so it should have no impact.
2. You only have one network card, so the sortnetif plug-in is meaningless.
3. 10ec:8168 is the rtl8168 network card. There are two drivers, r8168 and r8169_lk, in the current driver that can drive this network card.

In summary, please try to cancel the sortnetif plug-in and disable the r8169_lk module.

Gurabity, 2023-11-26:
As you suggested, I tried removing the module and add-on.
Try it after the update in version 11.4 and try it by creating a new USB.
I tried removing only r8169_lk from the module, or both r8169_lk and r8169, but it didn't work...
Should I really only use it on 11.4?
[logs.zip](https://github.com/wjz304/rr/files/13466986/logs.zip)

Gurabity, 2023-11-26:
Just in case, I used both the basic and 4G versions of the latest version of rr, and used two 16G USBs alternately.

wjz304, 2023-11-26:
> Try to roll back the driver version separately to the driver 11.5 that comes with RR 11.4. https://github.com/wjz304/rr-modules/releases/download/23.11.5/modules.zip
> 
> Download it, connect to RR via ssh and upload in "menu -> update menu -> Local upload" rebuild and reboot

If it work in this way, it seems to confirm that it is an issue with ASPM.

wjz304, 2024-03-30:
复现请重新打开

closed 2024-03-30 by wjz304