RROrg/rr #386: Вопрос по обновлению.

longtongue91, 2023-11-26:
Собственно, как правильно обновляться. Иногда обновление встают адекватно, иногда приходится переустанавливать через загрузчик. Недавно на минорном обновлении update 1 система сломалась, а тут на update 3 обновилась. если какие-то правила и рекомендации по обновлению?

wjz304, 2023-11-26:
Can you tell me more about "система сломалась" ?

longtongue91, 2023-11-26:
> система сломалась
Я не правильно выразился=) Действительно автоматически загрузился загрузучик.

wjz304, 2023-11-26:
> Планируете подробный гайд по обновлениям?
Not yet, [image]

Because it actually has strong fault tolerance, it is possible to update DSM or RR first. There are many ambiguous words in Chinese, and if I write in Chinese, the translation will be difficult to understand, but my English is also very poor,

I will consider this matter