RROrg/rr #391: BUG SA6400 自定义MAC地址设置不成功

onlypeng, 2023-11-28:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  23.11.10
 * addons:  23.11.18
 * modules:  23.11.11
 * lkms:  23.10.0

DSM: 
 * model:  SA6400
 * version:  7.2

Issue:  设置Cmdline 中自定SN/MAC后在群晖中查看sn正确，mac不为自定义mac

logs:  无日志

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

Promix953, 2023-11-28:
[https://github.com/wjz304/rr/releases/tag/23.9.7](https://github.com/wjz304/rr/releases/tag/23.9.7)
读一下changelog

onlypeng, 2023-11-28:
也就是说群晖里面显示的是实际mac地址，实际激活使用的参数还是自己设置的mac

closed 2023-11-28 by onlypeng