RROrg/rr #393: BUG Advanced Media Extensions 提示：无法执行此操作

onlypeng, 2023-11-28:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org 04/01/2014
 * CPU:  I3-12100
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  23.11.10
 * addons:  23.11.18
 * modules:  23.11.11
 * lkms:  23.10.0

DSM: 
 * model:  SA6400
 * version:  7.2

Issue:  Advanced Media Extensions 提示：无法执行此操作

logs:  starting serial terminal on interface serial0
[ 3174.306921] audit: type=1400 audit(1701156910.281:114): apparmor="DENIED" operation="mount" info="failed srcname match" error=-13 profile="/usr/syno/sbin/synoscgi//SYNO.AME.License" name="/tmp/bootmnt/" pid=11322 comm="SYNO.AME.Licens" fstype="ext2" srcname="/dev/synoboot2" flags="rw, silent"
[ 3174.317378] audit: type=1400 audit(1701156910.281:115): apparmor="DENIED" operation="mount" info="failed srcname match" error=-13 profile="/usr/syno/sbin/synoscgi//SYNO.AME.License" name="/tmp/bootmnt/" pid=11322 comm="SYNO.AME.Licens" fstype="ext4" srcname="/dev/synoboot2" flags="rw, silent"
[ 3174.324821] audit: type=1400 audit(1701156910.281:116): apparmor="DENIED" operation="mount" info="failed srcname match" error=-13 profile="/usr/syno/sbin/synoscgi//SYNO.AME.License" name="/tmp/bootmnt/" pid=11322 comm="SYNO.AME.Licens" fstype="vfat" srcname="/dev/synoboot2" flags="rw, silent"
[ 3174.329663] audit: type=1400 audit(1701156910.281:117): apparmor="DENIED" operation="mount" info="failed srcname match" error=-13 profile="/usr/syno/sbin/synoscgi//SYNO.AME.License" name="/tmp/bootmnt/" pid=11322 comm="SYNO.AME.Licens" fstype="ext2" srcname="/dev/synoboot2" flags="rw, silent"
[ 3174.346390] audit: type=1400 audit(1701156910.281:118): apparmor="DENIED" operation="mount" info="failed srcname match" error=-13 profile="/usr/syno/sbin/synoscgi//SYNO.AME.License" name="/tmp/bootmnt/" pid=11322 comm="SYNO.AME.Licens" fstype="ext4" srcname="/dev/synoboot2" flags="rw, silent"
[ 3174.364186] audit: type=1400 audit(1701156910.281:119): apparmor="DENIED" operation="mount" info="failed srcname match" error=-13 profile="/usr/syno/sbin/synoscgi//SYNO.AME.License" name="/tmp/bootmnt/" pid=11322 comm="SYNO.AME.Licens" fstype="vfat" srcname="/dev/synoboot2" flags="rw, silent"

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2023-11-28:
引导盘挂载不上

wjz304, 2023-11-28:
没遇到过，需要更多日志看看

onlypeng, 2023-11-28:
这段日志是在打开Advanced Media Extensions应用时报错这段时间获取的日志
[image]

wjz304, 2023-11-28:
都说了没遇到过，
log的字面意思： /dev/synoboot2 的 srcname 不匹配 挂载失败，synoboot2 就是 引导盘的第二个分区，
其他就不知道了

onlypeng, 2023-11-28:
好的，麻烦了

closed 2023-11-28 by onlypeng