RROrg/rr #400: Promox Intel 12代 SR-IOV 虚拟 vGPU 直通jellyfin硬解成果

chchia, 2024-03-30:
首先感谢@wjz304

这个不是bug，算是个简易教程给需要的朋友看看。

我的系统 i5-12400， Proxmox 最新版本 6.5 kernel

1. BIOS打开SR-IOV

2. 按这个教程 https://www.derekseaman.com/2023/11/proxmox-ve-8-1-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html

3. 使用 RR 编译 SA6400 使用 custom kernel

4. Proxmox里添加虚拟显卡给SA6400
[image]

5. DSM里写个script每次开机写入权限
`#!/bin/sh
sudo chmod 666 /dev/dri/*`

6. 使用官方jellyfin docker，如果你有portainer最好简单很多，在portainer给jellyfin这几个权限。
[image]

7.  jellyfin里开启硬件解码
[image]

8. jellyfin播放视频时，可在proxmox主机看到硬件解码正常使用
[image]

daxiguaya, 2024-04-03:
感谢

hhgsh, 2024-04-04:
我的n305打开seiov后其他格式正常硬解，但不能解码杜比视界。老版本引导可以

wjz304, 2024-04-07:
https://github.com/RROrg/rr/issues/604

Foreverux, 2024-04-15:
你好，可以请教一个问题吗？
第三步是怎么弄的呀？我使用custom编译SA6400型号后安装没有  /dev/dri文件夹

chchia, 2024-04-15:
那你有在第四步把显卡直通给DSM吗？

NJcoxx, 2024-07-16:
OpenCL色调映射正常？
https://github.com/RROrg/rr/issues/1222

zzdywc, 2024-08-09:
OpenCL色调映射依旧存在问题，不管是jellyfin还是emby，都会出现错误，jellyfin那边似乎是出现图像破碎（moetayuko/intel-gpu-i915-backports#24），我使用emby解码播放时，会打开播放器，但是没有任何图像显示，过一会提示未兼容的流退出播放，查看解码日志如下：
```
>> ThrottleBySegmentRequest: Latest request position unknown
>> ThrottleBySegmentRequest: Latest request position unknown
>> ThrottleBySegmentRequest: Latest request position unknown
>> ThrottleBySegmentRequest: Latest request position unknown
>> ThrottleBySegmentRequest: Latest request position unknown
```

NJcoxx, 2024-08-09:
> OpenCL色调映射依旧存在问题，不管是jellyfin还是emby，都会出现错误，jellyfin那边似乎是出现图像破碎（[moetayuko/intel-gpu-i915-backports#24](https://github.com/moetayuko/intel-gpu-i915-backports/issues/24)），我使用emby解码播放时，会打开播放器，但是没有任何图像显示，过 [...]

大概率是SR-IOV驱动的问题，可以试试strongtz的驱动，他的已经更新了

zzdywc, 2024-08-09:
> 大概率是SR-IOV驱动的问题，可以试试strongtz的驱动，他的已经更新了

哥们你回复的真快啊，但是我用的是moetayuko的驱动，切换起来感觉有点麻烦，我试试换他的驱动

NJcoxx, 2024-08-09:
> 哥们你回复的真快啊，但是我用的是moetayuko的驱动，切换起来感觉有点麻烦，我试试换他的驱动

因为我正在看帖，pve的驱动好换，主要是要换DSM内的驱动，i卡老老实实用vpp吧

zzdywc, 2024-08-09:
> 大概率是SR-IOV驱动的问题，可以试试strongtz的驱动，他的已经更新了

而且我看另外一条strongtz的issue（strongtz/i915-sriov-dkms#57），这个问题还是open状态

wjz304, 2024-08-24:
https://github.com/RROrg/rr/issues/868