RROrg/rr #521: 联想X1S屏幕支持，求指导

doa2011, 2024-04-02:
大佬，首先感谢你的付出，让引导越来越趋于完美。
之前关注过联想的X1S，做工非常的优秀，但系统很垃圾。虽然在用8100，非常的完美，但耐不住生命在于折腾，前段时间入手了一个，白黑连夜的研究，现在sa6400下所有硬件都工作良好，转码也很职称了，除了那个led小屏幕，试着去查了下原机系统屏幕显示的实现方式和系统内核，感觉是屏幕的这个驱动没有加载，想尝试编译下i2c_smbus和 i2c_i801模块，但失败了。
编译平台按网上教程，在 epyc7002开发者平台上进行编译会失败，在ubuntu上可以编译成功但无法使用。
看了咸鱼上有大佬已经搞定了屏幕显示，求大佬和基友支招，是否我的思路是错误的，编译驱动的做法不可取么？
以下是X1S 原机 系统lsmod：
```
root@lnase515:~# lsmod
Module                  Size  Used by
xt_MASQUERADE          20480  2
iptable_nat            16384  1
nf_nat                 45056  2 iptable_nat,xt_MASQUERADE
xt_state               16384  0
xt_conntrack           16384  1
nf_conntrack          147456  4 xt_conntrack,nf_nat,xt_state,xt_MASQUERADE
nf_defrag_ipv6         24576  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
xt_tcpudp              20480  20
iptable_filter         16384  1
bpfilter               16384  0
bonding               167936  0
i915                 2252800  1
rw_bios_nv             16384  0
drm_kms_helper        233472  1 i915
cec                    53248  2 drm_kms_helper,i915
mei_hdcp               24576  0
x86_pkg_temp_thermal    20480  0
intel_powerclamp       20480  0
crct10dif_pclmul       16384  1
intel_lpss_pci         20480  0
crc32_pclmul           16384  0
ghash_clmulni_intel    16384  0
intel_lpss             16384  1 intel_lpss_pci
aesni_intel           372736  0
raid1                  45056  1
igc                   114688  0
mei_me                 40960  1
crypto_simd            16384  1 aesni_intel
cryptd                 24576  2 crypto_simd,ghash_clmulni_intel
i2c_algo_bit           24576  1 i915
glue_helper            16384  1 aesni_intel
intel_cstate           24576  0
idma64                 20480  0
fb_sys_fops            16384  1 drm_kms_helper
processor_thermal_device    24576  0
mei                   118784  3 mei_hdcp,mei_me
i2c_i801               32768  0
syscopyarea            16384  1 drm_kms_helper
drm                   528384  3 drm_kms_helper,i915
int340x_thermal_zone    16384  1 processor_thermal_device
sysfillrect            16384  1 drm_kms_helper
video                  53248  1 i915
acpi_tad               16384  0
i2c_smbus              20480  1 i2c_i801
sysimgblt              16384  1 drm_kms_helper
efi_pstore             16384  0
pcspkr                 16384  0
virt_dma               20480  1 idma64
intel_soc_dts_iosf     20480  1 processor_thermal_device
acpi_pad              184320  0
ecryptfs              106496  0
ip_tables              28672  2 iptable_filter,iptable_nat
x_tables               49152  6 xt_conntrack,iptable_filter,xt_state,xt_tcpudp,ip_tables,xt_MASQUERADE
```
以下是编译时出错：
```
root@fbelavenuto-syno-compiler-1:/input# /opt/do.sh compile-module epyc7002
Compiling module for epyc7002-5.10.55...
make: Entering directory '/opt/epyc7002/build'
  CC [M]  /tmp/input/busses/i2c-i801.o
  CC [M]  /tmp/input/algos/i2c-algo-bit.o
  MODPOST /tmp/input/Module.symvers
ERROR: modpost: "check_signature" [/tmp/input/busses/i2c-i801.ko] undefined!
scripts/Makefile.modpost:124: recipe for target '/tmp/input/Module.symvers' fail                                     ed
make[1]: *** [/tmp/input/Module.symvers] Error 1
make[1]: *** Deleting file '/tmp/input/Module.symvers'
Makefile:1726: recipe for target 'modules' failed
make: *** [modules] Error 2
make: Leaving directory '/opt/epyc7002/build'
```
以下是在Ubuntu上编译加载时出错信息：
sa6400 insmod:
```
[ 798.049451] i2c_smbus: Unknown symbol __dynamic_dev_dbg (err -2)
[ 798.055913] i2c_smbus: Unknown symbol __stack_chk_fail (err -2)
[ 812.174900] i2c_i801: Unknown symbol acpi_has_watchdog (err -2)
[ 812.181285] i2c_i801: Unknown symbol i2c_register_spd (err -2)
[ 812.187529] i2c_i801: Unknown symbol __dynamic_dev_dbg (err -2)
[ 812.193863] i2c_i801: Unknown symbol __stack_chk_fail (err -2)
[ 812.200104] i2c_i801: Unknown symbol pv_ops (err -2)
[ 812.205428] i2c_i801: Unknown symbol check_signature (err -2)
```
附件是原机系统screen相关文件：
[screen.zip](https://github.com/RROrg/rr/files/14830609/screen.zip)

wjz304, 2024-04-02:
ERROR: modpost: "check_signature" [/tmp/input/busses/i2c-i801.ko] undefined!

不知道你这是咋编的，check_signature 第三方都去掉了

wjz304, 2024-04-02:
acpi_has_watchdog 也不好搞

wjz304, 2024-04-13:
无进一步反馈，本issue关闭

closed 2024-04-13 by wjz304

ogierJACK, 2024-12-13:
/sys/class/gpio/gpio332，在原厂系统里面这个文件的内容是啥？

ogierJACK, 2024-12-13:
/sys/class/gpio/gpio332/value，在原厂系统里面这个文件的内容是啥？