RROrg/rr #582: BUG

tinysun, 2024-04-03:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  To Be Filled By O.E.M. To Be Filled By O.E.M./Z390 Phantom Gaming 7, BIOS P1.30 11/29/2019
 * CPU:  Intel(R) Core(TM) i9-9900KF CPU @ 3.60GHz
 * NIC:  eth1(e1000e)  eth0(r8125): NOT CONNECTED

RR version: (You can find it in the update menu.)  
 * RR:  rr-24.3.9
 * addons:  默认未修改
 * modules: 默认未修改  
 * lkms:  默认未修改

DSM: 
 * model:  :DS3622xs+ 918+ 等等都试过了 都是相同的问题
 * version:  7.2(69057)

Issue:  
安装啥的都没问题，但是在引导启动的时候，引导不起来，直接重启 查看日志发现以下问题。好像和现实有关？（我用ARPL能引导，但是无法安装）我使用的amd卡。
logs:  
kern.warn kernel: [   11.504171] uvesafb: probe of uvesafb.0 failed with error -22
(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-04-07:
warn 无关紧要

wjz304, 2024-04-11:
无进一步反馈，该issue关闭

closed 2024-04-11 by wjz304