RROrg/rr #641: BUG mac address NOT applied

realizm, 2024-04-04:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  intel N100
 * NIC:  4 * intel i226

RR version: (You can find it in the update menu.)  
 * RR:  24.3.9
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  ds920+
 * version:  7.2

Issue:  

mac address NOT applied.

userconfig.yml
<pre><code>
mac1: "001132e****1"
mac2: "001132e****2"
mac3: "00d0b4024ee2"
mac4: "00d0b4024ee3"
</code></pre>

logs:  
[logs.zip](https://github.com/RROrg/rr/files/14872883/logs.zip)
[image]

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-04-04:
https://github.com/RROrg/rr/releases/tag/23.9.7

realizm, 2024-04-04:
Thanks for the quick reply

closed 2024-04-04 by realizm