RROrg/rr #654: BUG RR SA6400 7.2最新版本无法使用jellyfin进行色调映射

CWWK-TechnicalSupport, 2024-04-05:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Default string Default string/Default string, BIOS 5.27 03/14/2024
 * CPU:  Intel(R) N100
 * NIC:  eth0(igc)

RR version: (You can find it in the update menu.)  
 * RR:  24.4.0
 * addons:  24.4.0
 * modules:  24.4.0
 * lkms:  24.4.0

DSM: 
 * model:  SA6400
 * version:  7.2(69057u4)

Issue:  RR V24.4.0  SA6400 无法使用jellyfin进行DV色调映射

logs:  以下是nyanmisaka/jellyfin:latest的日志

/videos/f6d9a560-e3d2-35a1-ab99-c86cbf289f06/hls1/main/0.ts

{"Protocol":0,"Id":"f6d9a560e3d235a1ab99c86cbf289f06","Path":"/media/Download/\u6D4B\u8BD5\u89C6\u9891/dolby-vision-amaze.mp4","EncoderPath":null,"EncoderProtocol":null,"Type":0,"Container":"mov,mp4,m4a,3gp,3g2,mj2","Size":199246498,"Name":"dolby-vision-amaze","IsRemote":false,"ETag":"ac1c202c889dd7adc703e86fa834c89d","RunTimeTicks":562000000,"ReadAtNativeFramerate":false,"IgnoreDts":false,"IgnoreIndex":false,"GenPtsInput":false,"SupportsTranscoding":true,"SupportsDirectStream":true,"SupportsDirectPlay":true,"IsInfiniteStream":false,"RequiresOpening":false,"OpenToken":null,"RequiresClosing":false,"LiveStreamId":null,"BufferMs":null,"RequiresLooping":false,"SupportsProbing":true,"VideoType":0,"IsoType":null,"Video3DFormat":null,"MediaStreams":[{"Codec":"hevc","CodecTag":"dvhe","Language":"und","ColorRange":null,"ColorSpace":null,"ColorTransfer":null,"ColorPrimaries":null,"DvVersionMajor":1,"DvVersionMinor":0,"DvProfile":5,"DvLevel":9,"RpuPresentFlag":1,"ElPresentFlag":0,"BlPresentFlag":1,"DvBlSignalCompatibilityId":0,"Comment":null,"TimeBase":"1/60000","CodecTimeBase":null,"Title":null,"VideoRange":"HDR","VideoRangeType":"DOVI","VideoDoViTitle":"DV Profile 5","LocalizedUndefined":null,"LocalizedDefault":null,"LocalizedForced":null,"LocalizedExternal":null,"DisplayTitle":"4K HEVC HDR","NalLengthSize":null,"IsInterlaced":false,"IsAVC":null,"ChannelLayout":null,"BitRate":27713921,"BitDepth":10,"RefFrames":1,"PacketLength":null,"Channels":null,"SampleRate":null,"IsDefault":true,"IsForced":false,"Height":2160,"Width":3840,"AverageFrameRate":60,"RealFrameRate":60,"Profile":"Main 10","Type":1,"AspectRatio":"16:9","Index":0,"Score":null,"IsExternal":false,"DeliveryMethod":null,"DeliveryUrl":null,"IsExternalUrl":null,"IsTextSubtitleStream":false,"SupportsExternalStream":false,"Path":null,"PixelFormat":"yuv420p10le","Level":153,"IsAnamorphic":null},{"Codec":"eac3","CodecTag":"ec-3","Language":"und","ColorRange":null,"ColorSpace":null,"ColorTransfer":null,"ColorPrimaries":null,"DvVersionMajor":null,"DvVersionMinor":null,"DvProfile":null,"DvLevel":null,"RpuPresentFlag":null,"ElPresentFlag":null,"BlPresentFlag":null,"DvBlSignalCompatibilityId":null,"Comment":null,"TimeBase":"1/48000","CodecTimeBase":null,"Title":"sound handler","VideoRange":null,"VideoRangeType":null,"VideoDoViTitle":null,"LocalizedUndefined":null,"LocalizedDefault":null,"LocalizedForced":null,"LocalizedExternal":null,"DisplayTitle":"sound handler - Dolby Digital\u002B - 5.1 - Default","NalLengthSize":null,"IsInterlaced":false,"IsAVC":null,"ChannelLayout":"5.1","BitRate":640000,"BitDepth":null,"RefFrames":null,"PacketLength":null,"Channels":6,"SampleRate":48000,"IsDefault":true,"IsForced":false,"Height":null,"Width":null,"AverageFrameRate":null,"RealFrameRate":null,"Profile":null,"Type":0,"AspectRatio":null,"Index":1,"Score":null,"IsExternal":false,"DeliveryMethod":null,"DeliveryUrl":null,"IsExternalUrl":null,"IsTextSubtitleStream":false,"SupportsExternalStream":false,"Path":null,"PixelFormat":null,"Level":0,"IsAnamorphic":null}],"MediaAttachments":[],"Formats":[],"Bitrate":28362490,"Timestamp":null,"RequiredHttpHeaders":{},"TranscodingUrl":null,"TranscodingSubProtocol":null,"TranscodingContainer":null,"AnalyzeDurationMs":null,"DefaultAudioStreamIndex":null,"DefaultSubtitleStreamIndex":null}

/usr/lib/jellyfin-ffmpeg/ffmpeg -analyzeduration 200M -probesize 1G -init_hw_device vaapi=va:,kernel_driver=i915,driver=iHD -init_hw_device qsv=qs@va -init_hw_device opencl=ocl@va -filter_hw_device qs -hwaccel vaapi -hwaccel_output_format vaapi -autorotate 0 -i file:"/media/Download/测试视频/dolby-vision-amaze.mp4" -autoscale 0 -map_metadata -1 -map_chapters -1 -threads 0 -map 0:0 -map 0:1 -map -0:s -codec:v:0 h264_qsv -preset veryfast -look_ahead 0 -b:v 9616000 -maxrate 9616000 -bufsize 19232000 -profile:v:0 high -level 51 -g:v:0 240 -keyint_min:v:0 240 -vf "setparams=color_primaries=bt2020:color_trc=smpte2084:colorspace=bt2020nc,scale_vaapi=w=1920:h=1080:extra_hw_frames=24,hwmap=derive_device=opencl,tonemap_opencl=format=nv12:p=bt709:t=bt709:m=bt709:tonemap=bt2390:peak=100:desat=0,hwmap=derive_device=qsv:reverse=1:extra_hw_frames=16,format=qsv" -codec:a:0 libfdk_aac -ac 2 -ab 384000 -af "volume=2" -copyts -avoid_negative_ts disabled -max_muxing_queue_size 2048 -f hls -max_delay 5000000 -hls_time 4 -hls_segment_type mpegts -start_number 0 -hls_segment_filename "/config/transcodes/bb84dbf3f6d720f9821b403fe4ffe34c%d.ts" -hls_playlist_type vod -hls_list_size 0 -y "/config/transcodes/bb84dbf3f6d720f9821b403fe4ffe34c.m3u8"

ffmpeg version 6.0.1-Jellyfin Copyright (c) 2000-2023 the FFmpeg developers
built with gcc 12 (Debian 12.2.0-14)
configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-static --disable-libxcb --disable-sdl2 --disable-xlib --enable-lto --enable-gpl --enable-version3 --enable-shared --enable-gmp --enable-gnutls --enable-chromaprint --enable-opencl --enable-libdrm --enable-libass --enable-libfreetype --enable-libfribidi --enable-libfontconfig --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvorbis --enable-libopenmpt --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libfdk-aac --arch=amd64 --enable-libshaderc --enable-libplacebo --enable-vulkan --enable-vaapi --enable-amf --enable-libvpl --enable-ffnvcodec --enable-cuda --enable-cuda-llvm --enable-cuvid --enable-nvdec --enable-nvenc
libavutil 58. 2.100 / 58. 2.100
libavcodec 60. 3.100 / 60. 3.100
libavformat 60. 3.100 / 60. 3.100
libavdevice 60. 1.100 / 60. 1.100
libavfilter 9. 3.100 / 9. 3.100
libswscale 7. 1.100 / 7. 1.100
libswresample 4. 10.100 / 4. 10.100
libpostproc 57. 1.100 / 57. 1.100
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_21
libva info: va_openDriver() returns 0
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_21
libva info: va_openDriver() returns 0
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'file:/media/Download/测试视频/dolby-vision-amaze.mp4':
Metadata:
major_brand : mp42
minor_version : 1
compatible_brands: mp42dby1isom
creation_time : 2017-04-13T20:09:18.000000Z
Duration: 00:00:56.20, start: 0.000000, bitrate: 28362 kb/s
Stream #0:0[0x1](https://github.com/MoetaYuko/intel-gpu-i915-backports/issues/und): Video: hevc (Main 10) (dvhe / 0x65687664), yuv420p10le(tv), 3840x2160 [SAR 1:1 DAR 16:9], 27713 kb/s, 60 fps, 60 tbr, 60k tbn (default)
Metadata:
creation_time : 2017-04-13T20:09:18.000000Z
handler_name : video handler
vendor_id : [0][0][0][0]
encoder : DOVI Coding
Side data:
DOVI configuration record: version: 1.0, profile: 5, level: 9, rpu flag: 1, el flag: 0, bl flag: 1, compatibility id: 0
Stream #0:1[0x2](https://github.com/MoetaYuko/intel-gpu-i915-backports/issues/und): Audio: eac3 (ec-3 / 0x332D6365), 48000 Hz, 5.1(side), fltp, 640 kb/s (default)
Metadata:
creation_time : 2017-04-13T20:09:18.000000Z
handler_name : sound handler
vendor_id : [0][0][0][0]
Side data:
audio service type: main
Stream mapping:
[... 140 lines cut ...]
[ 32.740282] i915 display info: has_fpga_dbg: yes
[ 32.772491] i915 display info: has_gmch: no
[ 32.777192] i915 display info: has_hotplug: yes
[ 32.782228] i915 display info: has_hti: no
[ 32.786797] i915 display info: has_ipc: yes
[ 32.791490] i915 display info: has_overlay: no
[ 32.796494] i915 display info: has_psr: yes
[ 32.801153] i915 display info: has_psr_hw_tracking: no
[ 32.806852] i915 display info: overlay_needs_physical: no
[ 32.812931] i915 display info: supports_tv: no
[ 32.817896] i915 display info: has_hdcp: yes
[ 32.822671] i915 display info: has_dmc: yes
[ 32.827350] i915 display info: has_dsc: yes

以及 uname -a
Linux nas 5.10.55+ #69057 SMP Fri Jan 12 17:02:57 CST 2024 x86_64 GNU/Linux synology_epyc7002_sa6400

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined! 

@MoetaYuko

MoetaYuko, 2024-04-05:
在群晖里用 docker 高权限运行 archlinux，安装 [intel-compute-runtime](https://archlinux.org/packages/?name=intel-compute-runtime) 和 [clinfo](https://archlinux.org/packages/?name=clinfo)，`clinfo` 正常输出无报错：
```
[root@archlinux-1 /]# clinfo
Number of platforms                               1
  Platform Name                                   Intel(R) OpenCL Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 3.0
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_byte_addressable_store cl_khr_device_uuid cl_khr_fp16 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_command_queue_families cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_driver_diagnostics cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_intel_subgroups_char cl_intel_subgroups_long cl_khr_il_program cl_intel_mem_force_host_memory cl_khr_subgroup_extended_types cl_khr_subgroup_non_uniform_vote cl_khr_subgroup_ballot cl_khr_subgroup_non_uniform_arithmetic cl_khr_subgroup_shuffle cl_khr_subgroup_shuffle_relative cl_khr_subgroup_clustered_reduce cl_intel_device_attribute_query cl_khr_suggested_local_work_size cl_intel_split_work_group_barrier cl_intel_spirv_media_block_io cl_intel_spirv_subgroups cl_khr_spirv_linkonce_odr cl_khr_spirv_no_integer_wrap_decoration cl_intel_unified_shared_memory cl_khr_mipmap_image cl_khr_mipmap_image_writes cl_ext_float_atomics cl_khr_external_memory cl_intel_planar_yuv cl_intel_packed_yuv cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_3d_image_writes cl_intel_media_block_io cl_intel_subgroup_local_block_io cl_khr_integer_dot_product cl_khr_gl_sharing cl_khr_gl_depth_images cl_khr_gl_event cl_khr_gl_msaa_sharing cl_intel_sharing_format_query cl_khr_pci_bus_info
  Platform Extensions with Version                cl_khr_byte_addressable_store                                    0x400000 (1.0.0)
                                                  cl_khr_device_uuid                                               0x400000 (1.0.0)
                                                  cl_khr_fp16                                                      0x400000 (1.0.0)
                                                  cl_khr_global_int32_base_atomics                                 0x400000 (1.0.0)
                                                  cl_khr_global_int32_extended_atomics                             0x400000 (1.0.0)
                                                  cl_khr_icd                                                       0x400000 (1.0.0)
                                                  cl_khr_local_int32_base_atomics                                  0x400000 (1.0.0)
                                                  cl_khr_local_int32_extended_atomics                              0x400000 (1.0.0)
                                                  cl_intel_command_queue_families                                  0x400000 (1.0.0)
                                                  cl_intel_subgroups                                               0x400000 (1.0.0)
                                                  cl_intel_required_subgroup_size                                  0x400000 (1.0.0)
                                                  cl_intel_subgroups_short                                         0x400000 (1.0.0)
                                                  cl_khr_spir                                                      0x400000 (1.0.0)
                                                  cl_intel_accelerator                                             0x400000 (1.0.0)
                                                  cl_intel_driver_diagnostics                                      0x400000 (1.0.0)
                                                  cl_khr_priority_hints                                            0x400000 (1.0.0)
                                                  cl_khr_throttle_hints                                            0x400000 (1.0.0)
                                                  cl_khr_create_command_queue                                      0x400000 (1.0.0)
                                                  cl_intel_subgroups_char                                          0x400000 (1.0.0)
                                                  cl_intel_subgroups_long                                          0x400000 (1.0.0)
                                                  cl_khr_il_program                                                0x400000 (1.0.0)
                                                  cl_intel_mem_force_host_memory                                   0x400000 (1.0.0)
                                                  cl_khr_subgroup_extended_types                                   0x400000 (1.0.0)
                                                  cl_khr_subgroup_non_uniform_vote                                 0x400000 (1.0.0)
                                                  cl_khr_subgroup_ballot                                           0x400000 (1.0.0)
[... cut ...]
| PCIe   Bandwidth (send                 )                          9.72 GB/s |
| PCIe   Bandwidth (   receive           )                          9.75 GB/s |
| PCIe   Bandwidth (        bidirectional)            (Gen4 x16)    8.18 GB/s |
|-----------------------------------------------------------------------------|
|-----------------------------------------------------------------------------|
| Done. Press Enter to exit.                                                  |
'-----------------------------------------------------------------------------'
```

不像是驱动有问题的样子

CWWK-TechnicalSupport, 2024-04-05:
> 在群晖里用 docker 高权限运行 archlinux，安装 [intel-compute-runtime](https://archlinux.org/packages/?name=intel-compute-runtime) 和 [clinfo](https://archlinux.org/packages/?name=clinfo)，`clinfo` 正常输出无报错：
>
> ``` [...]

要不然你和nyanmisaka大佬沟通一下看问题在哪里？

MoetaYuko, 2024-04-05:
cc @nyanmisaka

nyanmisaka, 2024-04-05:
> cc @nyanmisaka

跟OP交流的结果是更新了内核驱动这边之后才出的问题，之前的旧版本是一切正常的。

还有这位，也是同样的问题。https://github.com/RROrg/rr/issues/400#issuecomment-2036009391

近期对i915 KMD那边进行过很大的变动么？我们这边在实体机mainline上测试都是正常的。

MoetaYuko, 2024-04-05:
> > cc @nyanmisaka
> 
> 跟OP交流的结果是更新了内核驱动这边之后才出的问题，之前的旧版本是一切正常的。
> 
> 还有这位，也是同样的问题。[#400 (comment)](https://github.com/RROrg/rr/issues/400#issuecomment-2036009391)
> 
> 近期对i915 KMD那边进行过很大的变动么？我们这边在实体机mainline上测试都是正常的。

之前很多人反馈过 i915 驱动更换后 qsv 下色调映射故障（vaapi 正常），但同时贴出来的内核日志没有任何报错，ffmpeg 也跟本帖一样只返回一个错误码，所以我完全看不懂哪里可能会有问题。

是否存在 jellyfin 打包的用户空间 media driver 跟内核驱动不兼容的可能性？这个故障怎么脱离 jellyfin 环境，单独使用 ffmpeg 进行复现？

nyanmisaka, 2024-04-05:
> 之前很多人反馈过 i915 驱动更换后 qsv 下色调映射故障（vaapi 正常），但同时贴出来的内核日志没有任何报错，ffmpeg 也跟本帖一样只返回一个错误码，所以我完全看不懂哪里可能会有问题。
>
> 是否存在 jellyfin 打包的用户空间 media driver 跟内核驱动不兼容的可能性？这个故障怎么脱离 jellyfin 环境，单独使用 ffmpeg 进行复现？

jellyfin打包的media driver iHD是和linux主线内核保持兼容性，以及有着和各大主要linux发行版类似的构建参数。但DG1这个特殊型号需要PROD OOT内核，因此目前没有启用支持。这个帖子里的N100和N305和DG1就更没啥关系了。你也可以删除`/usr/lib/jellyfin-ffmpeg/lib/dri/*_drv_video.so`来使用自己构建的media driver。

> [Parsed_tonemap_opencl_3 @ 0x5605617f4940] Failed to enqueue kernel: -6.

OP的环境是N100 5.10内核。OP给出的报错代码-6查询得到为`CL_OUT_OF_HOST_MEMORY` https://gist.github.com/bmount/4a7144ce801e5569a0b6

借用OP日志中的命令行，用[jellyfin-ffmpeg6 deb包](https://github.com/jellyfin/jellyfin-ffmpeg/releases/tag/v6.0.1-5)对杜比视界Profile 5的视频进行色调映射并重新编码。测试视频（/path/to/dolby-vision-amaze.mp4）可以从[这个链接](https://www.demolandia.net/4k-video-test/dolby-vision/page-1.html)中的第一个下载到。

`
/usr/lib/jellyfin-ffmpeg/ffmpeg -init_hw_device vaapi=va:,kernel_driver=i915,driver=iHD -init_hw_device qsv=qs@va -init_hw_device opencl=ocl@va -filter_hw_device qs -hwaccel vaapi -hwaccel_output_format vaapi -autorotate 0 -i file:"/path/to/dolby-vision-amaze.mp4" -autoscale 0 -map_metadata -1 -map_chapters -1 -threads 0 -map 0:0 -map 0:1 -map -0:s -codec:v:0 h264_qsv -preset veryfast -look_ahead 0 -b:v 9616000 -maxrate 9616000 -bufsize 19232000 -profile:v:0 high -level 51 -g:v:0 240 -keyint_min:v:0 240 -vf "setparams=color_primaries=bt2020:color_trc=smpte2084:colorspace=bt2020nc,scale_vaapi=w=1920:h=1080:extra_hw_frames=24,hwmap=derive_device=opencl,tonemap_opencl=format=nv12:p=bt709:t=bt709:m=bt709:tonemap=bt2390:peak=100:desat=0,hwmap=derive_device=qsv:reverse=1:extra_hw_frames=16,format=qsv" -an -sn -f null -
`

MoetaYuko, 2024-04-05:
```
[root@archlinux-1 ~]# /usr/lib/jellyfin-ffmpeg/ffmpeg -init_hw_device vaapi=va:,kernel_driver=i915,driver=iHD -init_hw_device qsv=qs@va -init_hw_device opencl=ocl@va -filter_hw_device qs -hwaccel vaapi -hwaccel_output_formatvaapi -autorotate 0 -i file:"dolby-vision-amaze-(dolby-vision)-(www.demolandia.net).mp4" -autoscale 0 -map_metadata -1 -map_chapters -1 -threads 0 -map 0:0 -map 0:1 -map -0:s -codec:v:0 h264_qsv -preset veryfast -look_ahead 0 -b:v 9616000 -maxrate 9616000 -bufsize 19232000 -profile:v:0 high -level 51 -g:v:0 240 -keyint_min:v:0 240 -vf "setparams=color_primaries=bt2020:color_trc=smpte2084:colorspace=bt2020nc,scale_vaapi=w=1920:h=1080:extra_hw_frames=24,hwmap=derive_device=opencl,tonemap_opencl=format=nv12:p=bt709:t=bt709:m=bt709:tonemap=bt2390:peak=100:desat=0,hwmap=derive_device=qsv:reverse=1:extra_hw_frames=16,format=qsv" -an -sn -f null -
ffmpeg version 6.0.1-Jellyfin Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13.2.1 (GCC) 20230801
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-shared --disable-libxcb --disable-sdl2 --disable-xlib --enable-gpl --enable-version3 --enable-static --enable-gmp --enable-gnutls --enable-chromaprint --enable-libfontconfig --enable-libass --enable-libbluray --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libmp3lame --enable-libopus --enable-libopenmpt --enable-libtheora --enable-libvorbis --enable-libdav1d --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libshaderc --enable-libplacebo --enable-vulkan --enable-opencl --enable-vaapi --enable-amf --enable-libvpl --enable-ffnvcodec --enable-cuda --enable-cuda-llvm --enable-cuvid --enable-nvdec --enable-nvenc
  libavutil      58.  2.100 / 58.  2.100
  libavcodec     60.  3.100 / 60.  3.100
  libavformat    60.  3.100 / 60.  3.100
  libavdevice    60.  1.100 / 60.  1.100
  libavfilter     9.  3.100 /  9.  3.100
  libswscale      7.  1.100 /  7.  1.100
  libswresample   4. 10.100 /  4. 10.100
  libpostproc    57.  1.100 / 57.  1.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'file:dolby-vision-amaze-(dolby-vision)-(www.demolandia.net).mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 1
    compatible_brands: mp42dby1isom
    creation_time   : 2017-04-13T20:09:18.000000Z
  Duration: 00:00:56.20, start: 0.000000, bitrate: 28362 kb/s
  Stream #0:0[0x1](und): Video: hevc (Main 10) (dvhe / 0x65687664), yuv420p10le(tv), 3840x2160 [SAR 1:1 DAR 16:9], 27713 kb/s, 60 fps, 60 tbr, 60k tbn (default)
    Metadata:
      creation_time   : 2017-04-13T20:09:18.000000Z
      handler_name    : video handler
      vendor_id       : [0][0][0][0]
      encoder         : DOVI Coding
    Side data:
      DOVI configuration record: version: 1.0, profile: 5, level: 9, rpu flag: 1, el flag: 0, bl flag: 1, compatibility id: 0
  Stream #0:1[0x2](und): Audio: eac3 (ec-3 / 0x332D6365), 48000 Hz, 5.1(side), fltp, 640 kb/s (default)
    Metadata:
      creation_time   : 2017-04-13T20:09:18.000000Z
      handler_name    : sound handler
      vendor_id       : [0][0][0][0]
    Side data:
      audio service type: main
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (native) -> h264 (h264_qsv))
Press [q] to stop, [?] for help
Abort was called at 987 line in file:
/usr/src/debug/intel-compute-runtime/compute-runtime-23.48.27912.11/shared/source/os_interface/linux/drm_memory_manager.cpp
Aborted (core dumped)
```
```
[3639961.185869] i915 0000:02:00.0: [drm:drm_ioctl [drm]] comm="ffmpeg" pid=27222, dev=0xe280, auth=1, I915_GEM_CONTEXT_SETPARAM
[3639961.187163] i915 0000:02:00.0: [drm:drm_ioctl [drm]] comm="ffmpeg", pid=27222, ret=-22
```
这个 ioctl 返回 -EINVAL，我后面有空了加 log 跟一下

CWWK-TechnicalSupport, 2024-04-07:
> ```
> [root@archlinux-1 ~]# /usr/lib/jellyfin-ffmpeg/ffmpeg -init_hw_device vaapi=va:,kernel_driver=i915,driver=iHD -init_hw_device qsv=qs@va -init_hw_device opencl=ocl@va -filter_hw_device qs [...]

有结果吗？

wjz304, 2024-04-07:
https://github.com/RROrg/rr/issues/604

hhgsh, 2024-04-12:
这么多大佬，，膜拜！感谢！但小白我没看懂该怎么解决。。。

closed 2024-04-14 by MoetaYuko

MoetaYuko, 2024-04-14:
故障原因是由于公共 api 行为变化，从高版本内核移植到 5.10 的 dma-buf 向用户空间返回的文件描述符没有设置 `FMODE_LSEEK` 标志位，导致用户空间 Intel(R) Graphics Compute Runtime 无法通过 `lseek` 系统调用查询对应 buf 的大小，进而触发 abort。下一版 rr 将修复该问题

hhgsh, 2024-04-14:
> 故障原因是由于公共 api 行为变化，从高版本内核移植到 5.10 的 dma-buf 向用户空间返回的文件描述符没有设置 `FMODE_LSEEK` 标志位，导致用户空间 Intel(R) Graphics Compute Runtime 无法通过 `lseek` 系统调用查询对应 buf 的大小，进而触发 abort。下一版 rr 将修复该问题

感谢大佬辛勤付出！

CWWK-TechnicalSupport, 2024-04-15:
> 故障原因是由于公共 api 行为变化，从高版本内核移植到 5.10 的 dma-buf 向用户空间返回的文件描述符没有设置 `FMODE_LSEEK` 标志位，导致用户空间 Intel(R) Graphics Compute Runtime 无法通过 `lseek` 系统调用查询对应 buf 的大小，进而触发 abort。下一版 rr 将修复该问题

谢谢大佬的修复