RROrg/rr #702: BUG：SA6400搭配MCX311A无法驱动

zrwd01, 2024-04-07:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * CPU:  D1581
 * NIC:  MCX311A

RR version: (You can find it in the update menu.)  
 * RR: 24.4.1 
 * addons:  cpuinfo，hdddb，nvmesystem，nvmevolume，storagepanel，
 * modules:  
 * lkms:  

DSM: 
 * model:  all
 * version:  最新

Issue:  
SA6400搭配MCX311A无法驱动
logs:  
[image]

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-04-08:
vid pid

wjz304, 2024-04-11:
无进一步反馈，该issue关闭

closed 2024-04-11 by wjz304