RROrg/rr #718: BUG ESXI 7 or 8 直通SATA控制器后无法引导

naoki66, 2024-04-08:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI: 7.0 Update 3 和 8.0 Update 2
 * CPU:  10100T
 * NIC:  VMNET3

RR version: (You can find it in the update menu.)  
 * RR:  24.4.1
 * addons:  
  acpid: ""
  mountloader: ""
  reboottoloader: ""
  addincards: ""
  codecpatch: ""
  console: ""
  cpuinfo: ""
  dbgutils: ""
  expands: ""
  hdddb: ""
  i915le10th: ""
  setrootpw: ""
  synoconfbkp: 5 sybk
  updatenotify: ""
  storagepanel: TOWER_8_Bay 1X2
  powersched: ""

 *modules:
  8139cp: ""
  9p: ""
  9pnet: ""
  9pnet_virtio: ""
  aacraid: ""
  adm1021: ""
  adm9240: ""
  aic94xx: ""
  alx: ""
  apex: ""
  aqc111: ""
  asix: ""
  atl1c: ""
  atl1e: ""
  atlantic: ""
  auxiliary: ""
  ax88179_178a: ""
  b44: ""
  backlight: ""
  bcm-phy-lib: ""
  be2net: ""
  bitblit: ""
  bnx2: ""
  bnx2x: ""
  bnx2x_mod: ""
  bnxt_en: ""
  broadcom: ""
  btcoexist: ""
  button: ""
  cdc_ether: ""
  cdc_ncm: ""
  cdrom: ""
  cfbcopyarea: ""
  cfbfillrect: ""
  cfbimgblt: ""
  cfg80211: ""
  ch: ""
  ch341: ""
  cls_fw: ""
  cls_u32: ""
  cn: ""
  cp210x: ""
  cpufreq_conservative: ""
  cpufreq_governor: ""
  cpufreq_ondemand: ""
  crc-itu-t: ""
  cxgb: ""
  cxgb3: ""
  cxgb4: ""
  cxgb4vf: ""
  dm-log: ""
  dm-mirror: ""
  dm-mod: ""
  dm-multipath: ""
  dm-queue-length: ""
  dm-region-hash: ""
  dm-round-robin: ""
  dm-service-time: ""
  dm9601: ""
  dme1737: ""
  drm: ""
  DRM: ""
  drm_kms_helper: ""
  drm_panel_orientation_quirks: ""
  dvb-core: ""
  dvb-usb-dib0700: ""
  dvb-usb-dvbsky: ""
  dvb-usb: ""
  dvb_usb_v2: ""
  e1000: ""
  e1000e: ""
  eeprom_93cx6: ""
  efifb: ""
  ehci-hcd: ""
  ehci-pci: ""
  f71882fg: ""
  f75375s: ""
  fb: ""
  fb_sys_fops: ""
  fbcon: ""
  fbcon_ccw: ""
  fbcon_cw: ""
  fbcon_rotate: ""
  fbcon_ud: ""
  fbdev: ""
  font: ""
  ftdi_sio: ""
  gasket: ""
  hpsa: ""
  hvc_console: ""
  hwmon-vid: ""
  i40e: ""
  i915: ""
  iavf: ""
  ib_addr: ""
  ib_cm: ""
  ib_core: ""
  ib_mad: ""
  ib_sa: ""
  ib_umad: ""
  ib_uverbs: ""
  ice: ""
  igb: ""
  igbvf: ""
  igc: ""
  intel_auxiliary: ""
  iosf_mbi: ""
  ipv6: ""
  irqbypass: ""
  iscsi_tcp: ""
  it87: ""
  iw_cm: ""
  iwldvm: ""
  iwlmvm: ""
  iwlwifi: ""
  ixgb: ""
  ixgbe: ""
  ixgbevf: ""
  jc42: ""
  jme: ""
  kvm-amd: ""
  kvm-intel: ""
  kvm: ""
  lib80211: ""
  libiscsi: ""
  libiscsi_tcp: ""
  libphy: ""
  libsas: ""
  llc: ""
  lm75: ""
  lm78: ""
  lm90: ""
  lm95245: ""
  m88ds3103: ""
  mac80211: ""
  mdio: ""
  megaraid: ""
  megaraid_mbox: ""
  megaraid_mm: ""
  megaraid_sas: ""
  mii: ""
  mlx4_core: ""
  mlx4_en: ""
  mlx4_ib: ""
  mlx5_core: ""
  mlx5_ib: ""
  mlxsw_core: ""
  mlxsw_pci: ""
  mmc_block: ""
  mmc_core: ""
  mpt3sas: ""
  mptbase: ""
  mptctl: ""
  mptsas: ""
  mptscsih: ""
  mptspi: ""
  mt7601u: ""
  mtk-sd: ""
  mvsas: ""
  mwifiex: ""
  mwifiex_sdio: ""
  mwifiex_usb: ""
  nct6775: ""
  netxen_nic: ""
  p8022: ""
  phy-mt65xx-usb3: ""
  phy: ""
  pmbus: ""
  pmbus_core: ""
  prism2_usb: ""
  processor: ""
  psnap: ""
  qcaspi: ""
  qed: ""
  qede: ""
  qla3xxx: ""
  qlcnic: ""
  qlge: ""
  r8125: ""
  r8125_v9: ""
  r8126: ""
  r8152: ""
  r8168: ""
  r8169: ""
  r8169_lk: ""
  raid_class: ""
  rc-core: ""
  realtek: ""
  regmap-i2c: ""
  rndis_host: ""
  rtc-cmos: ""
  rtl8187: ""
  rtl8188ee: ""
  rtl818x_pci: ""
  rtl8192c-common: ""
  rtl8192ce: ""
  rtl8192cu: ""
  rtl8192de: ""
  rtl8192ee: ""
  rtl8192se: ""
  rtl8723-common: ""
  rtl8723ae: ""
  rtl8723be: ""
  rtl8821ae: ""
  rtl8xxxu: ""
  rtl_pci: ""
  rtl_usb: ""
  rtlwifi: ""
  rtsx_pci: ""
  rtsx_pci_sdmmc: ""
  rtsx_usb: ""
  rtsx_usb_sdmmc: ""
  sch_htb: ""
  sch_netem: ""
  sch_sfq: ""
  scsi_transport_sas: ""
  scsi_transport_spi: ""
  sdhci-acpi: ""
  sdhci-pci-data: ""
  sdhci-pci: ""
  sdhci-pltfm: ""
  sdhci: ""
  sdio_uart: ""
  sg: ""
  si2157: ""
  si2168: ""
  skge: ""
  sky2: ""
  softcursor: ""
  sp2: ""
  sr_mod: ""
  ssb: ""
  staging: ""
  stp: ""
  synobios: ""
  syscopyarea: ""
  sysfillrect: ""
  sysimgblt: ""
  tg3: ""
  thermal: ""
  tps65217: ""
  ts2020: ""
  uhci-hcd: ""
  usbnet: ""
  usbserial: ""
  ushc: ""
  vesafb: ""
  vga16fb: ""
  vgastate: ""
  via-sdmmc: ""
  video: ""
  virtio: ""
  virtio_blk: ""
  virtio_console: ""
  virtio_input: ""
  virtio_mmio: ""
  virtio_net: ""
  virtio_pci: ""
  virtio_ring: ""
  virtio_scsi: ""
  vmw_pvscsi: ""
  vmw_vmci: ""
  vmxnet3: ""
  vub300: ""
  w83781d: ""
  w83793: ""
  wireguard: ""
 
 * lkms:  prod

DSM: 
 * model:  920
 * version:  7.2 69057

Issue:  直通SATA控制器，启动失联，无法安装
[image]

logs:  
添加了dbgutils，但是依然没有logs，只好上传串口信息
[image]

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

naoki66, 2024-04-08:
[serial.zip](https://github.com/RROrg/rr/files/14900135/serial.zip)
串口信息压缩

naoki66, 2024-04-08:
失联前的最后一条信息是
Installing addon console - modules
[   15.720995] efifb: probing for efifb
[   15.721456] efifb: framebuffer at 0xf0000000, mapped to 0xffffc90000e00000, using 1408k, total 1406k
[   15.722456] efifb: mode is 800x600x24, linelength=2400, pages=352
[   15.723126] efifb: scrolling: redraw
[   15.723507] efifb: Truecolor: size=0:8:8:8, shift=0:16:8:0
[   15.724226] fb0: EFI VGA frame buffer device
Killed

wjz304, 2024-04-08:
取消 console addon ？

naoki66, 2024-04-09:
取消后问题依旧，暂时只能换pve了，然后没有问题。
另外能否提个意见，启动盘能否更新下pciids？便于lspci命令的时候更容易找到对应名称的设备地址。
能否增加 hwinfo工具，便于定位USB端口？现在DT文件用于群晖准确度很高。

wjz304, 2024-04-09:
pciids 默认的就可以了

有lshw，lsusb，...

wjz304, 2024-04-09:
也可以安装 opkg ，

closed 2024-04-13 by wjz304