RROrg/rr #739: R8152网卡机器在安装DSM系统成功后失联

infinitr0us, 2024-04-09:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Lenovo S940 笔记本
 * CPU:  Intel 10代 I7-1065G7
 * NIC:  USB千兆网卡R8152

RR version: (You can find it in the update menu.)  
 * RR:  24.4.1
 * addons:  添加了NVMEsystem
 * modules:  默认，无修改
 * lkms:  默认，无修改

DSM: 
 * model:  SA6400
 * version:  DSM 7.2.1-69057 update4

Issue:  黑裙新手，拿旧的笔记本实验装黑裙。整个引导创建和编译过程都没有任何问题，路由器那边可以识别到，安装界面也可以正常打开。但是一旦装完DSM系统之后进入那个10分钟读条，机器自动重启然后进入BOOT DSM选项之后就会失联，笔记本显示器上面显示是正常BOOT DSM，IP可以正常获得（但是不知道是不是直接沿用的编译引导时候保存的IP），但是路由器里面是等满了10分钟也看不到，IP也ping不通。希望得到大佬的帮助，帮忙看看是不是网卡（R8152）的问题。

logs:  直接ping黑裙的IP的话会返回“无法访问目标主机”。另外还有什么需要的log，我可以回头再开机试试。

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

infinitr0us, 2024-04-09:
可以在补充一点后续实验的发现：
1. 换成一个RTL8156的2.5G USB网卡之后成功启动进入了DSM一次，但是重启机器之后就再也连不上了，后续又尝试复现了十几次，但是都甚至没有启动到DSM一次
2. 从路由器自己观察了一下链接的情况，在DSM启动过程中，在十秒倒计时的时候（刚获取完IP后）是路由器中可见的，但是一道了后面load kernal 和boot DSM之后就断联了

wjz304, 2024-04-09:
usb 网卡有些会默认关闭

wjz304, 2024-04-09:
取log吧

infinitr0us, 2024-04-10:
> 取log吧

感谢大佬指点，最近有点忙，后续有时间我取log再发上来看看，我先把这个issue暂时关了

closed 2024-04-10 by infinitr0us