RROrg/rr #758: BUG vmm固件为UEFI的无法进入系统(包括ISO启动文件也不行)

jialuming, 2024-04-10:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  24.4.1
 * addons:  默认的
 * modules:  all
 * lkms:  

DSM: 
 * model:  ds3622xs+
 * version:  7.2

Issue:  UEFI固件时，windows也进不去了，windows原版安装盘也无法加载，legacy的可以，原arpl引导没有这些问题

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-04-11:
现有信息不足，无法调查

wjz304, 2024-04-25:
https://github.com/RROrg/rr/releases/tag/24.4.7
test

closed 2024-04-25 by wjz304