RROrg/rr #793: 制作U盘启动后，重启后一直黑屏，无法进入引导界面

pengfeigong88, 2024-04-14:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  J4125
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  rr-24.4.1.img
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  
制作U盘启动盘后，插入主机（J4125）迷你小主机后，进BIOS设置U盘启动后重启，一直黑屏，无法进入引导界面

pengfeigong88, 2024-04-14:
[image]
[image]
[image]
[image]
[image]

wjz304, 2024-04-14:
关闭 安全启动 和 快速启动

pengfeigong88, 2024-04-15:
关闭了也没用，最后是在bios South Bridge里把OS SELECTION 从windows改成了linux才可以了

closed 2024-04-19 by wjz304