RROrg/rr #800: BUG

wahabe, 2024-04-14:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  DSM version: 7.1 not work

[image]
[image]

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wahabe, 2024-04-14:
[image]

wahabe, 2024-04-14:
[image]

wjz304, 2024-04-19:
无法调查

closed 2024-04-19 by wjz304