RROrg/rr #807: BUG SA6400-7.1安装系统时不能识别硬盘

iziyou, 2024-04-15:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org 04/01/2014
 * CPU:  Intel(R) Xeon(R) E-2144G CPU @ 3.60GHz
 * NIC:  eth0(vmxnet3)、eth1(mlx5_core)

RR version: (You can find it in the update menu.)  
 * RR:  rr-24.4.2
 * addons:  acpid、mountloader、reboottoloader
 * modules:  默认（应该是全部modules）
 * lkms:  prod

DSM: 
 * model:  SA6400
 * version: 7.1(42962u2)

Issue:  PVE虚拟机环境，引导盘ide0, 虚拟硬盘sata0, 还有一张lsi 9400(接了两个硬盘)，SA6400-7.1安装系统时提示找不到硬盘

logs:

wjz304, 2024-04-20:
get logs

wjz304, 2024-04-25:
无进一步反馈，本issue关闭

closed 2024-04-25 by wjz304