RROrg/rr #816: 能否添加Mellanox 25GMCX4421A-ACQN驱动

timtown01, 2024-04-16:
v24.4.3版本添加了许多mellanox驱动，但经我测试该版本不支持Mellanox 25GMCX4421A-ACQN网卡，能否在后续版本中添加该网卡驱动，万分感谢！

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU: Intel(R) Xeon(R) Bronze 3104
 * NIC:  Mellanox 25GMCX4421A-ACQN

RR version: (You can find it in the update menu.)  
 * RR:  v24.4.3
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-04-16:
pidvid

timtown01, 2024-04-22:
> pidvid

15b3:1015

wjz304, 2024-04-22:
有

timtown01, 2024-04-24:
> 有
> 
> root@ING-PC:/mnt/c/Users/Ing/Downloads/modules-24.4.3/broadwellnk-4.4.302# modinfo mlx5_core.ko filename: /mnt/c/Users/Ing/Downloads/modules-24.4.3/broadwellnk-4.4.302/mlx5_core.ko version: 3.0-1 license: Dual BSD/GPL description: Mellanox Connect-IB, ConnectX-4 core driver author: Eli Cohen [eli@mellanox.com](mailto:eli@mellanox.com) srcversion: 58907809152ACC56F02A32F alias: pci:v000015B3d00001016sv_sd_bc_sc_i* alias: pci:v000015B3d00001015sv_sd_bc_sc_i* alias: pci:v000015B3d00001014sv_sd_bc_sc_i* alias: pci:v000015B3d00001013sv_sd_bc_sc_i* alias: pci:v000015B3d00001012sv_sd_bc_sc_i* alias: pci:v000015B3d00001011sv_sd_bc_sc_i* depends: vermagic: 4.4.302+ SMP mod_unload parm: debug_mask:debug mask: 1 = dump cmd data, 2 = dump cmd exec time, 3 = both. Default=0 (int) parm: prof_sel:profile selector. Valid range 0 - 2 (int)

[image]
能查到，但是找不到网卡

wjz304, 2024-04-24:
DSM 能识别么

timtown01, 2024-04-24:
> DSM 能识别么
还没装，目前就这个网卡，走的OCP主板直连，有办法解决吗？

timtown01, 2024-04-24:
> lspci -nnk

[image]

wjz304, 2024-04-24:
https://forums.developer.nvidia.com/t/mlx5-missing-registers-bar-error-code-19/253354

wjz304, 2024-04-24:
[image]
[image]
[image]

jjia'shang加上试试吧

timtown01, 2024-04-24:
> ![image](https://private-user-images.githubusercontent.com/5615843/325093369-537e6d7a-9400-4cc3-8683-d249160044b8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]
grub.cfg改了 还是报-19

wjz304, 2024-04-24:
自己搜吧，RR是的驱动都是官方默认的

timtown01, 2024-04-24:
> 自己搜吧，RR是的驱动都是官方默认的

好的，麻烦您了

closed 2024-05-05 by wjz304