RROrg/rr #823: BUG

koihano, 2024-04-17:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI: COLORFLY BATTLE-AX B365M-D V20
 * CPU:  i3-8100
 * NIC:  r8168

RR version: (You can find it in the update menu.)  
 * RR:  24.4.0 to 24.4.0
 * addons:  
 * modules:  deafult
 * lkms:  default

DSM: 
 * model:  SA6400(officail)
 * version:  7.2.1u5

Issue:  
24.4.4使用rr高级菜单【恢复已安装的DSM系统】功能失效，提示未找到系统分区(md0)，回滚到24.4.0正常

logs:none  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-04-17:
rrshell 下 lsblk -f

koihano, 2024-04-19:
> rrshell 下 lsblk -f

root@rr:/opt/rr# lsblk -f
NAME      FSTYPE            FSVER    LABEL     UUID                                   FSAVAIL FSUSE% MOUNTPOINTS
sda                                                                                                  
`-sda1    linux_raid_member 1.2      DS_Epyc:3 c1128a87-f1ea-9278-2e22-c7c275570266                  
  `-md127 LVM2_member       LVM2 001           JHiXB6-vn7A-D6Mf-OfEQ-l0Pc-eBbE-NGz5BU                
sdb                                                                                                  
|-sdb1    linux_raid_member 0.90.0             f355a48a-38e5-78fa-3017-a5a8c86610be                  
|-sdb2    linux_raid_member 0.90.0             04ba052d-113a-057d-3017-a5a8c86610be                  
`-sdb5    linux_raid_member 1.2      DS_NVR:2  6dd272ef-5fb8-a61a-688f-304add9ac798                  
  `-md126 LVM2_member       LVM2 001           mv2YE6-LvXZ-1u4A-HtO9-ZIKn-o4ZM-3ERiZ4                
sdc                                                                                                  
|-sdc1    vfat              FAT32    RR1       DA16-9EF2                                39.5M    20% /mnt/p1
|-sdc2    ext2              1.0      RR2       a8889315-1cac-4e9b-b0e1-d8a16195afd9     44.3M     0% /mnt/p2
`-sdc3    ext4              1.0      RR3       617a3aca-4b56-42d7-8558-54411b344a7d     27.1G     2% /mnt/p3
sdd                                                                                                  
|-sdd1    linux_raid_member 0.90.0             f355a48a-38e5-78fa-3017-a5a8c86610be                  
|-sdd2    linux_raid_member 0.90.0             04ba052d-113a-057d-3017-a5a8c86610be                  
`-sdd5    linux_raid_member 1.2      DS_NVR:2  6dd272ef-5fb8-a61a-688f-304add9ac798                  
  `-md126 LVM2_member       LVM2 001           mv2YE6-LvXZ-1u4A-HtO9-ZIKn-o4ZM-3ERiZ4                
sde                                                                                                  
|-sde1    linux_raid_member 0.90.0             f355a48a-38e5-78fa-3017-a5a8c86610be                  
|-sde2    linux_raid_member 0.90.0             04ba052d-113a-057d-3017-a5a8c86610be                  
`-sde5    linux_raid_member 1.2      DS_NVR:2  6dd272ef-5fb8-a61a-688f-304add9ac798                  
  `-md126 LVM2_member       LVM2 001           mv2YE6-LvXZ-1u4A-HtO9-ZIKn-o4ZM-3ERiZ4                
sdf                                                                                                  
|-sdf1    linux_raid_member 0.90.0             f355a48a-38e5-78fa-3017-a5a8c86610be                  
|-sdf2    linux_raid_member 0.90.0             04ba052d-113a-057d-3017-a5a8c86610be                  
`-sdf5    linux_raid_member 1.2      DS_NVR:2  6dd272ef-5fb8-a61a-688f-304add9ac798                  
  `-md126 LVM2_member       LVM2 001           mv2YE6-LvXZ-1u4A-HtO9-ZIKn-o4ZM-3ERiZ4                
sdg                                                                                                  
|-sdg1    linux_raid_member 0.90.0             f355a48a-38e5-78fa-3017-a5a8c86610be                  
|-sdg2    linux_raid_member 0.90.0             04ba052d-113a-057d-3017-a5a8c86610be                  
`-sdg5    linux_raid_member 1.2      DS_NVR:2  6dd272ef-5fb8-a61a-688f-304add9ac798                  
  `-md126 LVM2_member       LVM2 001           mv2YE6-LvXZ-1u4A-HtO9-ZIKn-o4ZM-3ERiZ4                
sdh                                                                                                  
|-sdh1    linux_raid_member 0.90.0             f355a48a-38e5-78fa-3017-a5a8c86610be                  
|-sdh2    linux_raid_member 0.90.0             04ba052d-113a-057d-3017-a5a8c86610be                  
`-sdh5    linux_raid_member 1.2      DS_NVR:2  6dd272ef-5fb8-a61a-688f-304add9ac798                  
  `-md126 LVM2_member       LVM2 001           mv2YE6-LvXZ-1u4A-HtO9-ZIKn-o4ZM-3ERiZ4                
sdi                                                                                                  
|-sdi1    linux_raid_member 0.90.0             f355a48a-38e5-78fa-3017-a5a8c86610be                  
|-sdi2    linux_raid_member 0.90.0             04ba052d-113a-057d-3017-a5a8c86610be                  
`-sdi5    linux_raid_member 1.2      DS_NVR:2  6dd272ef-5fb8-a61a-688f-304add9ac798                  
  `-md126 LVM2_member       LVM2 001           mv2YE6-LvXZ-1u4A-HtO9-ZIKn-o4ZM-3ERiZ4                
sdj                                                                                                  
|-sdj1    linux_raid_member 0.90.0             f355a48a-38e5-78fa-3017-a5a8c86610be                  
|-sdj2    linux_raid_member 0.90.0             04ba052d-113a-057d-3017-a5a8c86610be                  
`-sdj5    linux_raid_member 1.2      DS_NVR:2  6dd272ef-5fb8-a61a-688f-304add9ac798                  
  `-md126 LVM2_member       LVM2 001           mv2YE6-LvXZ-1u4A-HtO9-ZIKn-o4ZM-3ERiZ4                
sdk                                                                                                  
|-sdk1    linux_raid_member 0.90.0             f355a48a-38e5-78fa-3017-a5a8c86610be                  
|-sdk2    linux_raid_member 0.90.0             04ba052d-113a-057d-3017-a5a8c86610be                  
`-sdk5    linux_raid_member 1.2      DS_NVR:2  6dd272ef-5fb8-a61a-688f-304add9ac798                  
  `-md126 LVM2_member       LVM2 001           mv2YE6-LvXZ-1u4A-HtO9-ZIKn-o4ZM-3ERiZ4

wjz304, 2024-04-19:
升级完应立即重启，
恢复已安装的DSM系统 的问题 下个版本

wjz304, 2024-04-20:
24.4.5

closed 2024-04-20 by wjz304