RROrg/rr #844: 未知原因导致DSM点关机或重启均无效，只能qm stop强行关闭虚拟机

missing233, 2024-04-20:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  PROXMOX 7.4-3
 * CPU:  Intel Xeon D-1581
 * NIC:  VirtIO

RR version: (You can find it in the update menu.)  
 * RR:  24.4.5
 * addons:  default
 * modules:  default
 * lkms:  default

DSM: 
 * model:  SA6400
 * version:  7.2.1-69057 Update 4

Issue:  
刚装好DSM时正常，不知何时开始DSM无法关机或重启。DSM关闭后VNC显示仍卡在系统启动后的界面，只能手动在PVE里qm stop来强行关闭。

logs:  
关机/重启前的一段日志：
`kvm: exiting hardware virtualization`之后PVE虚拟机没有被自动关闭。只能qm stop
```
[  203.747574] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 10403
[  203.749055] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 7645
[  203.750468] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  203.810617] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 11550
[  203.812108] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 5024
[  203.813512] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  203.839458] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 12164
[  203.841138] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 3112
[  203.842574] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  203.860502] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 12820
[  203.861989] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 1692
[  203.863344] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  203.877511] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 13188
[  203.878885] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 873
[  203.880225] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  203.895562] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 13516
[  203.896950] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 218
[  203.898291] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  204.497967] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 15154
[  204.499356] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 0
[  204.500693] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  204.516327] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 15400
[  204.517681] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 655
[  204.518976] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  204.545740] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 15687
[  204.547067] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 1419
[  204.548385] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  204.576772] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 16055
[  204.578096] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 2348
[  204.579383] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  204.607834] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 16383
[  204.609139] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 3549
[  204.610424] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  204.636536] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 16711
[  204.637847] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 4751
[  204.639121] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  204.652023] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 17038
[  204.653346] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 5898
[  204.654636] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  204.675662] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 17366
[  204.676970] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 6881
[  204.678425] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  204.707014] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 17653
[  204.708531] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 7700
[  204.709815] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[... 925 lines cut ...]
[  414.068868] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 3658
[  414.070189] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  414.094372] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 20028
[  414.095773] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 2894
[  414.097089] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  414.111456] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 20315
[  414.112816] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 2348
[  414.114129] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  414.135001] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 20848
[  414.136343] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 1529
[  414.137651] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  414.154324] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 21421
[  414.155667] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 436
[  414.156961] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
[  414.174768] evbug: Event. Dev: input3, Type: 3, Code: 0, Value: 21544
[  414.176104] evbug: Event. Dev: input3, Type: 3, Code: 1, Value: 109
[  414.177375] evbug: Event. Dev: input3, Type: 0, Code: 0, Value: 0
```

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-04-22:
[  264.401984] kvm: exiting hardware virtualization

missing233, 2024-04-22:
> 正常不会，[ 264.401984] kvm: exiting hardware virtualization 的问题 目前不好解决，加个计划任务，关机的时候
> 
> ```
> rmmod kvm_intel
> rmmod kvm_amd
> rmmod kvm
> rmmod irqbypass
> ```
> 
> 试试吧

添加之后关机log是这样的，仍然卡死不能关机。
另外我按照此DG1教程修改过PVE：https://blog.kkk.rs/archives/28
不过当时应该是没有问题
````
[  115.446204] Synotify use 16384 event queue size
[  120.194996] Module [kvm_intel] is removed. 
[  120.212929] Module [kvm] is removed. 
[  120.229810] Module [irqbypass] is removed. 
[  125.998173] vfat: synoboot1 mounted, process=mount
[  127.189955] synoboot1 unmounted, process=umount
[  OK  ] Stopped Traffic Control.
         Stopping Traffic Control...
[  OK  ] Stopped hook on all packages ready.
         Stopping hook on all packages ready...
         Stopping BTRFS quick balance...
[  OK  ] Stopped Synology Log bootup done.
         Stopping Synology Log bootup done...
[  OK  ] Removed slice SecureSignin.slice.
[  OK  ] Stopped Notify system is detected with slow drive.
         Stopping Notify system is detected with slow drive...
[  OK  ] Stopped Check synorbd on system volume.
         Stopping Check synorbd on system volume...
[  OK  ] Removed slice HyperBackup-ED's slice.
[  OK  ] Stopped Adds correct CPU Info.
         Stopping Adds correct CPU Info...
[  OK  ] Stopped rr notify.
         Stopping rr notify...
[  OK  ] Removed slice Synology Cache Advisor.
[  OK  ] Stopped Dump synoplatform info.
         Stopping Dump synoplatform info...
[  OK  ] Stopped Reassemble Pstore log into /var/log/pstore/.
         Stopping Reassemble Pstore log into /var/log/pstore/...
         Stopping Synology daemon for monitoring smpkg...
         Stopping WS-Discovery Server...
[  OK  ] Stopped Expanded miscellaneous.
         Stopping Expanded miscellaneous...
         Stopping WD idle3 timer...
[  OK  ] Stopped Execute packages' scripts during the boot-up.
         Stopping Execute packages' scripts during the boot-up...
[  OK  ] Stopped Notify bundle drive update after bootup done.
         Stopping Notify bundle drive update after bootup done...
[  OK  ] Stopped Notification for firmware upgrade result.
         Stopping Notification for firmware upgrade result...
[  OK  ] Stopped Preload synopki data to cache.
         Stopping Preload synopki data to cache...
[  OK  ] Stopped Mount usb fs.
         Stopping Mount usb fs...
[  OK  ] Stopped Check the disk is ssd/hdd.
         Stopping Check the disk is ssd/hdd...
         Stopping Write back disk latency to DB...
[  OK  ] Stopped Syno Data Protection - Btrfs Snapshot Usage Task Clean.
         Stopping Syno Data Protection - Btrfs Snapshot Usage Task Clean...
[  OK  ] Stopped Fan debug check.
         Stopping Fan debug check...
         Stopping Synology daemon for monitoring disk latency....
[  OK  ] Stopped Check Switch's Firmware Settings and External NICs for FS3410.
         Stopping Check Switch's Firmware Se... and External NICs for FS3410...
         Stopping Set IP Conflict Detect...
[  OK  ] Stopped target PostgreSQL adapter target.
[... 698 lines cut ...]
[  165.020640] zram: Removed device: zram20
[  165.028239] printk: systemd-shutdow: 187 output lines suppressed due to ratelimiting
[  165.030963] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[  165.037519] zram: Removed device: zram21
[  165.039418] systemd-journald[7700]: Received SIGTERM from PID 1 (systemd-shutdow).
[  165.041466] audit: type=1335 audit(1713805553.632:19): pid=7700 uid=0 auid=4294967295 tty=(none) ses=4294967295 subj==unconfined comm="systemd-journal" exe="/usr/lib/systemd/systemd-journald" nl-mcgrp=1 op=disconnect res=1
[  165.053578] zram: Removed device: zram22
[  165.065568] zram: Removed device: zram23
[  175.039355] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[  175.045224] systemd-shutdown[1]: Sending SIGKILL to PID 7591 (inetd).
[  175.046794] systemd-shutdown[1]: Sending SIGKILL to PID 8183 (systemd-udevd).
[  175.049455] systemd-shutdown[1]: Unmounting file systems.
[  175.050806] systemd-shutdown[1]: Unmounting /config.
[  175.051992] systemd-shutdown[1]: Unmounting /tmp.
[  175.098927] EXT4-fs (md0): re-mounted. Opts: (null)
[  175.103939] EXT4-fs (md0): re-mounted. Opts: (null)
[  175.105259] EXT4-fs (md0): re-mounted. Opts: (null)
[  175.106547] systemd-shutdown[1]: All filesystems unmounted.
[  175.108009] systemd-shutdown[1]: Deactivating swaps.
[  175.109335] systemd-shutdown[1]: All swaps deactivated.
[  175.110695] systemd-shutdown[1]: Detaching loop devices.
systemd-shutdown: run synobios_uninit OK
[  176.206774] flashcache_syno: Reboot hook start
[  176.207666] flashcache_syno: Reboot hook finish
````

wjz304, 2024-04-22:
我也只遇到过kvm 的问题，而且根据资料 这是kernel 5 的历史遗留问题，

wjz304, 2024-04-22:
此外 最好 q35 host

missing233, 2024-04-22:
> 此外 最好 q35 host

使用的就是q35。。头疼 这个问题真不知道怎么解决了（

missing233, 2024-04-24:
> 此外 最好 q35 host

重装DSM仍然无法解决问题，下载了最新的rr.img替换了之前的就恢复正常了。不知道是之前的镜像哪里出现了问题。

missing233, 2024-04-25:
> 我也只遇到过kvm 的问题，而且根据资料 这是kernel 5 的历史遗留问题，

从24.4.6升级到24.4.7后再次出现此问题，卡在`kvm: exiting hardware virtualization`

wjz304, 2024-04-26:
cmdline 添加 acpi=force 试试

so1ve, 2024-05-06:
装一个QEMU Guest Tools试试

wjz304, 2024-06-23:
无进一步反馈，有问题请重新创建issue

closed 2024-06-23 by wjz304

knatwl, 2025-01-25:
我也遇到同样的问题，不过我把xe显卡设备移除就可以正常关机不会卡住了。另外，我是12代核显，我的i915显卡也没有正常加载。