RROrg/rr #891: BUG.pcieBUG

xuwenhong, 2024-04-26:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

xuwenhong, 2024-04-26:
udevadm info /dev/nvme3n
正常的nvme的地址是P: /devices/pci0000:00/0000:00:17.0/0000:13:00.0/nvme/nvme2/nvme2n1
不正常时是：P: /devices/pci0000:ae/0000:ae:02.0/0000:b1:00.0/0000:b2:0c.0/0000:b6:00.0/nvme/nvme3/nvme3n1，
而且大于6个nvme时就不正常，请检查是不是像以前超过26个硬盘就不正常的原因！

wjz304, 2024-04-26:
不是

wjz304, 2024-04-26:
pci路径和你主板有关