RROrg/rr #896: 网络只有百兆

pengfeigong88, 2024-04-27:
请填写以下信息.  
[image]
[image]
网络只有百兆，相同的网线在电脑上就是千兆
网卡信息：
[image]

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  J4125
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  24.4.4
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS920+
 * version:  DSM 7.2.1-69057 Update 4

wjz304, 2024-04-27:
有三个 8168 的驱动，试试吧。
另外通过ethtool 查看设置信息

pengfeigong88, 2024-04-27:
> 有三个 8168 的驱动，试试吧。 另外通过ethtool 查看设置信息
怎么弄，有教程吗？

pengfeigong88, 2024-04-27:
[image]

pengfeigong88, 2024-04-28:
@wjz304 
[image]

wjz304, 2024-04-28:
[image]

pengfeigong88, 2024-04-30:
试过了，还是不行

snailium, 2024-06-06:
换条网线试试？

wjz304, 2025-05-08:
新版本如有问题 ，请重新创建issue

closed 2025-05-08 by wjz304