RROrg/rr #936: BUG EMMC引导，DT机型，引导盘任然会被认为是USB SD卡

naoki66, 2024-05-06:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  MetaServer Zero
 * CPU:  5105
 * NIC:  i225

RR version: (You can find it in the update menu.)  
 * RR:  [24.5.1](https://github.com/RROrg/rr/releases/tag/24.5.1)
 * addons:  
   reboottoloader: ''
  expands: ''
  codecpatch: ''
  powersched: ''
  cpuinfo: ''
  hdddb: ''
  mountloader: ''
  acpid: ''
  sensors: ''
  storagepanel: ''
  updatenotify: ''
  synoconfbkp: ''
  remotefs: ''
  setrootpw: ''

 * modules:  
   8139cp: ''
  9p: ''
  9pnet: ''
  9pnet_virtio: ''
  aacraid: ''
  adm1021: ''
  adm1031: ''
  adm9240: ''
  adt7470: ''
  adt7475: ''
  aic94xx: ''
  alcor_pci: ''
  alx: ''
  apex: ''
  aqc111: ''
  asix: ''
  atkbd: ''
  atl1c: ''
  atl1e: ''
  atlantic: ''
  ax88179_178a: ''
  b44: ''
  bcm-phy-lib: ''
  be2net: ''
  blk-mq-virtio: ''
  bnx2: ''
  bnx2x: ''
  bnx2x_mod: ''
  bnxt_en: ''
  broadcom: ''
  btcoexist: ''
  button: ''
  cdc_ether: ''
  cdc_ncm: ''
  cfbcopyarea: ''
  cfbfillrect: ''
  cfbimgblt: ''
  cfg80211: ''
  ch: ''
  ch341: ''
  cls_fw: ''
  cls_u32: ''
  cn: ''
  coretemp: ''
  cp210x: ''
  cpufreq_performance: ''
  cqhci: ''
  crc-ccitt: ''
  crc-itu-t: ''
  crc8: ''
  dm-log: ''
  dm-mirror: ''
  dm-mod: ''
  dm-multipath: ''
  dm-queue-length: ''
  dm-region-hash: ''
  dm-round-robin: ''
  dm-service-time: ''
  dm9601: ''
  dmabuf: ''
  dme1737: ''
  drm: ''
  drm_buddy: ''
  drm_display_helper: ''
  drm_kms_helper: ''
  drm_mipi_dsi: ''
  drm_panel_orientation_quirks: ''
  e1000: ''
  e1000e: ''
  eeprom_93cx6: ''
  ehci-hcd: ''
  ehci-pci: ''
  evbug: ''
  evdev: ''
  exfat: ''
  f71882fg: ''
  f75375s: ''
  failover: ''
  fb: ''
  fb_sys_fops: ''
  fbdev: ''
  fixed_phy: ''
  ftdi_sio: ''
  gasket: ''
  hpsa: ''
  hvc_console: ''
  hwmon-vid: ''
  i2c-algo-bit: ''
  i2c-i801: ''
  i2c-smbus: ''
  i40e: ''
  i8042: ''
  i915-compat: ''
  i915: ''
  iavf: ''
  ib_cm: ''
  ib_core: ''
  ib_umad: ''
  ib_uverbs: ''
  igb: ''
  igbvf: ''
  igc: ''
  intel-gtt: ''
  intel-lpss: ''
  intel_auxiliary: ''
  ipv6: ''
  irqbypass: ''
  iscsi_tcp: ''
  it87: ''
  iw_cm: ''
  iwldvm: ''
  iwlmvm: ''
  iwlwifi: ''
  ixgbe: ''
  ixgbevf: ''
  jc42: ''
  kvm-amd: ''
  kvm-intel: ''
  kvm: ''
  lib80211: ''
  libarc4: ''
  libdes: ''
  libfc: ''
  libiscsi: ''
  libiscsi_tcp: ''
  libphy: ''
  libsas: ''
  libsha256: ''
  llc: ''
  lm75: ''
  lm78: ''
  lm90: ''
  lm95245: ''
  mac80211: ''
  mdio_devres: ''
  megaraid: ''
  megaraid_mbox: ''
  megaraid_mm: ''
  megaraid_sas: ''
  mii: ''
  mlx4_core: ''
  mlx4_en: ''
  mlx4_ib: ''
  mlx5_core: ''
  mlx5_ib: ''
  mlxfw: ''
  mlxsw_core: ''
  mlxsw_i2c: ''
  mlxsw_pci: ''
  mmc_block: ''
  mmc_core: ''
  mpi3mr: ''
  mpt3sas: ''
  mptbase: ''
  mptctl: ''
  mptsas: ''
  mptscsih: ''
  mptspi: ''
  mt76: ''
  mt7601u: ''
  mtk-sd: ''
  mvsas: ''
  nct6683: ''
  nct6775: ''
  net_failover: ''
  netxen_nic: ''
  of_mdio: ''
  p8022: ''
  phy-intel-lgm-emmc: ''
  phy-lgm-usb: ''
  phy: ''
  pl2303: ''
  pmbus: ''
  pmbus_core: ''
  prism2_usb: ''
  psnap: ''
  pwm-fan: ''
  qca_7k_common: ''
  qed: ''
  qede: ''
  qedr: ''
  qla3xxx: ''
  qlcnic: ''
  qlge: ''
  r8125: ''
  r8125_v9: ''
  r8126: ''
  r8152: ''
  r8168: ''
  r8169: ''
  r8169_lk: ''
  realtek: ''
  regmap-i2c: ''
  rndis_host: ''
  rtc-cmos: ''
  rtl8187: ''
  rtl8188ee: ''
  rtl818x_pci: ''
  rtl8192c-common: ''
  rtl8192ce: ''
  rtl8192cu: ''
  rtl8192de: ''
  rtl8192ee: ''
  rtl8192se: ''
  rtl8723-common: ''
  rtl8723ae: ''
  rtl8723be: ''
  rtl8821ae: ''
  rtl8xxxu: ''
  rtl_pci: ''
  rtl_usb: ''
  rtlwifi: ''
  rtsx_pci: ''
  rtsx_pci_sdmmc: ''
  rtsx_usb: ''
  rtsx_usb_sdmmc: ''
  rtw_8723d: ''
  rtw_8723de: ''
  rtw_8723ds: ''
  rtw_8723du: ''
  rtw_8821c: ''
  rtw_8821ce: ''
  rtw_8821cs: ''
  rtw_8821cu: ''
  rtw_8822b: ''
  rtw_8822be: ''
  rtw_8822bs: ''
  rtw_8822bu: ''
  rtw_8822c: ''
  rtw_8822ce: ''
  rtw_8822cs: ''
  rtw_8822cu: ''
  rtw_core: ''
  rtw_pci: ''
  rtw_sdio: ''
  rtw_usb: ''
  sch_htb: ''
  sch_netem: ''
  sch_sfq: ''
  scsi_transport_fc: ''
  scsi_transport_sas: ''
  scsi_transport_spi: ''
  sdhci-acpi: ''
  sdhci-pci-data: ''
  sdhci-pci: ''
  sdhci-pltfm: ''
  sdhci-xenon-driver: ''
  sdhci: ''
  sdio_uart: ''
  sg: ''
  skge: ''
  sky2: ''
  ssb: ''
  stp: ''
  synobios: ''
  syscopyarea: ''
  sysfillrect: ''
  sysimgblt: ''
  tg3: ''
  thermal: ''
  thunderbolt-net: ''
  thunderbolt: ''
  tps65217: ''
  ttm: ''
  uhci-hcd: ''
  usbnet: ''
  usbserial: ''
  ushc: ''
  vfio: ''
  via-sdmmc: ''
  virtio: ''
  virtio_blk: ''
  virtio_console: ''
  virtio_input: ''
  virtio_mmio: ''
  virtio_net: ''
  virtio_pci: ''
  virtio_ring: ''
  virtio_scsi: ''
  vmw_pvscsi: ''
  vmw_vmci: ''
  vmxnet3: ''
  vub300: ''
  w83781d: ''
  w83793: ''
  wireguard: ''
  xz_dec: ''
 * lkms:  dev

DSM: 
 * model:  SA6400
 * version:  7.2.1 --69057Update5

Issue:  
dms消息会有如下提示
[image]

但是外接设备看不到emmc设备
[image]

我是USB是端口是用DT手动写死的。理论上，是不是不在这些端口下的设备不应该被识别？
    usb_slot@1 {
      usb2 {
        usb_port ="1-2";
      };
      usb3 {
        usb_port ="1-1";
      };
    };
    usb_slot@2 {
      usb2 {
        usb_port ="1-4";
      };
      usb3 {
        usb_port ="1-8";
      };
    };   

logs:  
很遗憾，添加了dbgutils这个插件，但是依旧没有log
[image]

(请先看一下#173、#175、#226的内容)  
(Plz review the content of #173, #175, #226 first)  
...

... 如果你提供不了详细信息，那就等有缘人吧!
... If you can't provide detailed information, then wait for someone who is destined!

wjz304, 2024-05-06:
mmc 引导的 后遗症，每次开机都会弹一次那个通知

lemon66789, 2024-05-06:
我也出现了这个问题，和你一样的机子。情况一模一样。

tsingui, 2024-05-15:
usb引导多简单

016, 2024-05-16:
I got same error, with same machine and 24.5.1 too, will test 24.5.4 late, just hope this issue will be-fixed. thx for everything. just fix it ...

wjz304, 2024-05-17:
DSM shell
```
cp -f /usr/lib/udev/script/sdcard.sh /usr/lib/udev/script/sdcard.sh.bak
echo -en '#!/bin/sh\nexit 0\n' > /usr/lib/udev/script/sdcard.sh
```
and reboot

wjz304, 2024-05-17:
Who can give me feedback on the results

016, 2024-05-17:
@wjz304 I will test it later tonight, will feedback soon.

016, 2024-05-17:
@wjz304 it's working, the the notifications msg is gone, everything looks good. thx.

wjz304, 2024-05-17:
RR shell
```
sed -i 's/sdhci,sdhci_pci,sdhci_acpi/xxxx/' /opt/rr/boot.sh
```
boot to DSM

016, 2024-05-17:
@wjz304 我不明白什么是 RR shell 是机器启动前引导的 shell吗？ 有任何文档我可以学习下吗？

wjz304, 2024-05-17:
[image]

016, 2024-05-17:
@wjz304 working on it , will update soon.

lemon66789, 2024-05-17:
[image]
不知是不是我操作不对，键入以后没有反应，重启后问题还存在

wjz304, 2024-05-17:
> DSM shell
> 
> ```
> cp -f /usr/lib/udev/script/sdcard.sh /usr/lib/udev/script/sdcard.sh.bak
> echo -en '#!/bin/sh\nexit 0\n' > /usr/lib/udev/script/sdcard.sh
> ```
> 
> and reboot

@lemon66789  这两个要一起

016, 2024-05-17:
@wjz304 I cannot see any changes, here is what I did.

1. reboot, select "Configure loader" to start "rr shell"
2. run CMD "sed -i 's/sdhci,sdhci_pci,sdhci_acpi/xxxx/' /opt/rr/boot.sh" //nothing output see pic1
3. use menu.sh go back menu, select "b 启动" //see pic2
4. after DSM start, no msg got. 

ps:
1. 之前的cp 和 echo的命令行已经生效了，所以这里我没看出任何差别。
2. 我的rr版本是24.5.1，如果有需要我可以安装24.5.4
[image]
[image]

lemon66789, 2024-05-17:
吃完饭用第一个代码试了试，确实解决了，作者大大霸气！

wjz304, 2024-05-19:
https://github.com/RROrg/rr/releases/tag/24.5.5

closed 2024-05-19 by wjz304